realsenseai/realsense-ros #3318: d435i sync in virtual machine (VMware)

autumntang, 2025-03-17:
Hello!

I have some questions regarding time synchronization for the D435i and am looking forward to your guidance. I am currently trying to implement hardware synchronization for the D435i. I have successfully achieved external synchronization for the depth frames using the following program, and it runs perfectly on Windows.

However, I now need to implement it on Ubuntu since I will later need to call related nodes in ROS. The issue is that I am using an Ubuntu virtual machine (VMware), and I keep encountering the following error when executing profile = pipeline.start(config): RuntimeError: xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error

Is there any way to skip this line of code while still achieving the same functionality?
-----------------------------------------------------------------------------------------------------------------------------
import pyrealsense2 as rs
import time

# 初始化 RealSense Pipeline 和 Config
pipeline = rs.pipeline()
config = rs.config()

# 配置深度流：分辨率 1280x720，格式 Z16，原生帧率 30Hz（触发上限 15Hz）
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)

# 检查设备连接
ctx = rs.context()
devices = ctx.query_devices()
if not devices:
    raise RuntimeError("未检测到 RealSense 设备！")
device = devices[0]

# 配置 Genlock 模式（模式4：单帧触发）
depth_sensor = device.first_depth_sensor()
depth_sensor.set_option(rs.option.inter_cam_sync_mode, 4)  # 模式4 = Genlock 单帧触发
# depth_sensor.set_option(rs.option.enable_auto_exposure, 0)  # 关闭自动曝光·
# depth_sensor.set_option(rs.option.exposure, 1000)           # 手动曝光时间（μs）

# 启动流传输
profile = pipeline.start(config)
print("D435i 已进入 Genlock 模式，等待外部 10Hz 触发信号...")

# 验证同步模式
sync_mode = depth_sensor.get_option(rs.option.inter_cam_sync_mode)
print(f"当前同步模式: {sync_mode} (4=Genlock 单帧触发)")

# 帧率与时间戳检测
frame_count = 0
start_time = time.time()
last_timestamp = None  # 记录上一帧硬件时间戳

try:
    while True:
        # 等待深度帧
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        current_timestamp = depth_frame.get_timestamp()  # 单位：毫秒

        # 新增：打印当前帧时间戳
        print(f"当前深度帧时间戳: {current_timestamp:.3f} ms")

        # 首次跳过时间戳计算
        if last_timestamp is None:
            last_timestamp = current_timestamp
            continue

        # 计算帧间隔（毫秒）
        frame_interval = current_timestamp - last_timestamp
        last_timestamp = current_timestamp
        frame_count += 1

        # 每10帧打印统计信息
        if frame_count % 10 == 0:
            elapsed_time = time.time() - start_time
            avg_fps = frame_count / elapsed_time
            print(f"平均帧率: {avg_fps:.2f} Hz | 最近帧间隔: {frame_interval:.1f} ms")

        # 检查帧间隔是否在预期范围（10Hz → 100ms ±10%容差）
        if not (90 < frame_interval < 110):
            print(f"警告！帧间隔异常: {frame_interval:.1f} ms")

except KeyboardInterrupt:
    print("用户中断...")
finally:
    pipeline.stop()
    print("数据流已停止")

MartyG-RealSense, 2025-03-18:
Hi @autumntang  Because the error is an **input / output** error, that suggests that the problem is related to the USB system.

It is possible that a configuration of 1280x720 and 30 FPS may be too much for a virtual machine, where the USB controllers are emulated.  You could test this by reducing resolution to 848x480 or the FPS to 15.

autumntang, 2025-03-18:
Thank you so much, It worked!

MartyG-RealSense, 2025-03-18:
That's great news - thanks very much for the update about your success!  :)

autumntang, 2025-03-19:
@MartyG-RealSense Sorry, I encountered some issues again. When configuring only the depth stream, it works when I lower the resolution. However, when I add the RGB stream, it stops working, even though I have reduced both the resolution and FPS of the RGB stream.
```python
config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 15)

MartyG-RealSense, 2025-03-19:
What happens if you change your Inter Cam Sync Mode value to '3' (Full Slave mode) instead of 4.  

In mode 3, depth and RGB are meant to be able to be synced, something that is not usually possible with the hardware sync system.  In truth mode 3 does not work very well, as it was intended for use with camera models like D415 that have an integrated RGB sensor.  However, it would be worth trying value 3 to see what happens with your D435i.

autumntang, 2025-03-20:
@MartyG-RealSense After I set the mode to 3, the camera fails to operate according to the 10Hz external pulse I provided, and the output timestamps are as follows. Although I am currently externally synchronizing the depth frames, my ultimate goal is for the RGB frames to work based on my external pulse. Therefore, I can only select the timestamp closest to the depth frame to ensure that the RGB frame aligns with the external pulse as much as possible. However, in the virtual machine, I am unable to configure the RGB stream using profile = pipeline.start(config), which is really frustrating for me. Can you give me some suggestions?

当前深度帧时间戳: 1742454872321.913 ms
警告！帧间隔异常: 66.8 ms
当前深度帧时间戳: 1742454872455.371 ms
警告！帧间隔异常: 133.5 ms
当前深度帧时间戳: 1742454872488.749 ms
警告！帧间隔异常: 33.4 ms
当前深度帧时间戳: 1742454872555.445 ms
警告！帧间隔异常: 66.7 ms
当前深度帧时间戳: 1742454872622.133 ms
警告！帧间隔异常: 66.7 ms
当前深度帧时间戳: 1742454872655.441 ms
警告！帧间隔异常: 33.3 ms
当前深度帧时间戳: 1742454872722.110 ms
警告！帧间隔异常: 66.7 ms
当前深度帧时间戳: 1742454872888.729 ms
警告！帧间隔异常: 166.6 ms

MartyG-RealSense, 2025-03-20:
It is not normally possible to sync depth and RGB streams of the same camera using hardware sync.  Mode 3 was an attempt to do this but as mentioned above, it did not work well.  The entire genlock sync system is no longer supported by Intel (as it was an experimental 'proof of concept' tool), with modes 1 and 2 being the only modes that technical support is officially provided for now, though genlock modes 3+continue to be accessible in the RealSense SDK.

What happens if you use 30 FPS for both depth and color but use 640x480 resolution for both streams?

```
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)
```

autumntang, 2025-03-20:
Error: RuntimeError: xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error

MartyG-RealSense, 2025-03-31:
Hi @autumntang  Do you require further assistance with this case, please?  Thanks!

autumntang, 2025-04-09:
Thanks for your suggestions! however, I must use d435i in VM for some reasons. I can use d435i well in VM by 'roslaunch rs_camera.launch',  but why 'config.enable_stream' can't. Is there any way to achieve external synchronization by roslaunch? @MartyG-RealSense

autumntang, 2025-04-09:
Moreover, Why I can't set RGB 1920*1080, 'Given stream configuration is not supported by the device!  Stream: Color, Stream Index: 0, Width: 1920, Height: 1080, FPS: -1, Format: None'

MartyG-RealSense, 2025-04-09:
You can enable hardware sync in ROS using the **inter_cam_sync_mode** parameter.  For example:

`roslaunch realsense2_camera rs_camera.launch inter_cam_sync_mode:=2`

closed 2025-05-01 by MartyG-RealSense