realsenseai/realsense-ros #3321: D455 RGB + IMU configuration for outdoor visual-inertial SLAM (ROS Noetic, ros1-legacy)

lyc1140889141, 2025-03-24:
Hi RealSense team, thank you for maintaining this great project!

I'm using Ubuntu 20.04 with ROS Noetic and working with the `ros1-legacy` branch of the RealSense ROS wrapper. I'm currently configuring the D455 for outdoor RGB + IMU recording for use in visual-inertial SLAM systems (e.g., VINS-Mono), and I have a few questions regarding best practices and recommended settings.
### 1. RGB Camera -> Controls settings in RealSense Viewer:
Under Auto Exposure controls, there are:
- `Auto Exposure Priority`
- `Global Time Enabled`
- `Backlight Compensation`
Which of these should be enabled for outdoor SLAM recording?
**My understanding:**
- `Auto Exposure Priority`: should be **disabled**, since it reduces framerate to gain exposure, which can break IMU-camera time alignment.
- `Global Time Enabled`: should be **enabled**, to ensure consistent timestamping between sensors (camera & IMU).
- `Backlight Compensation`: should be **disabled**, as it may introduce unstable exposure in high-contrast outdoor environments.
### 2. What exactly does `Global Time Enabled` do?
Does it make all sensors (e.g., RGB and IMU) use the same clock for timestamps, and does it affect the timestamps stored in rosbag that are later used for RGB-IMU alignment in VIO systems?
### 3. Should I enable `enable_sync` in the launch file?
I'm only using RGB and IMU streams (not depth or pointcloud). From what I understand, enabling `enable_sync` delays the RGB image to wait for other streams, which may **harm** time alignment with IMU.
Should I explicitly set:
```xml
<arg name="enable_sync" default="false"/>
```
### 4. Question about `unite_imu_method` and IMU frequency combinations
I tested and found that the D455 supports the following IMU configurations:
- Accel: up to 200Hz
- Gyro: up to 400Hz
This allows two commonly used setups:
- `accel=200`, `gyro=200` with `unite_imu_method:=copy`
- `accel=200`, `gyro=400` with `unite_imu_method:=linear_interpolation`
**According to the `ros1-legacy` branch's README:**
- `copy` avoids interpolation but may cause timestamp mismatch when gyro runs faster than accel
- `linear_interpolation` provides smoother timestamps but uses synthetic accel values
For generating the `/camera/imu` topic, which configuration yields lower timestamp and data error when compared to the original accel and gyro streams?
Specifically, does `accel=200 + gyro=200 + unite_imu_method=copy` produce more accurate timestamps and sensor values, or is `accel=200 + gyro=400 + unite_imu_method=linear_interpolation` better?

Any official recommendation for VIO use cases would be really helpful.
Thanks a lot in advance for your time and support!

[image]

MartyG-RealSense, 2025-03-24:
Hi @lyc1140889141  Thanks very much for your questions.

I would recommend starting with the default settings to see how well they work with your project and not begin changing settings until you have a problem.

1.  You can set global_time_enabled to false if you experience timing problems, as ROS has its own timing sync mechanism.

https://github.com/IntelRealSense/realsense-ros/issues/1581 is an example of a ROS1 case when images were being greatly delayed when global_time_enabled was true.

Setting auto_exposure_priority to false can help to maintain a constant FPS when both depth and color are enabled if one of the streams cannot provide the defined FPS (for example, it has been set to 30 FPS but can only provide 15).  If depth is not enabled though then you should not need to change this setting to false.

You should not usually have to make any changes to Backlight Compensation.

2.  **global_time_enabled** enables global timestamping, generating a common timestamp for all streams.  Information about the global time feature can be found at https://github.com/IntelRealSense/librealsense/pull/3909 and https://github.com/IntelRealSense/realsense-ros/issues/796

As far as I am aware it would not affect the values stored in a rosbag.

3. As you are not using depth, I would recommend having **enable_sync** on false.

4.  In most cases using the **linear_interpolation** setting for unite_imu_method is fine.  The 'copy' setting can produce more stable IMU messages if the IMU data is fluctuating a lot, though it is not necessarily more accurate than linear_interpolation.

In regard to VIO ... my knowledge of VIO is admittedly limited.  However, my research of your question indicates that RealSense ROS users who have used VIO have typically preferred copy over linear_interpolation as they felt the copy mode gave better stability. An example of this opinion from a VINS-Fusion user is here:

http://official-rtab-map-forum.206.s1.nabble.com/RTAB-map-with-vins-Fusion-tp6111p6824.html

A VINS-Mono user also set it to Copy.

lyc1140889141, 2025-03-26:
Hi, thanks again for your help!

1. Regarding **Backlight Compensation**, should it be enabled or disabled by default?  
   I previously adjusted this setting, so I am unsure whether the default value is `enabled` or `disabled`.

2. Before performing IMU calibration using `imu_utils` or `allan_variance_ros`, I checked the IMU readings in RealSense Viewer.  
   - In one specific orientation, the **g-norm** is around **9.58 m/s²**.  
   - In all other orientations, it fluctuates between **9.7 - 9.9 m/s²**.  
   
   In this case, do you recommend running `rs-imu-calibration.py` before proceeding with IMU noise modeling?

Thanks in advance for your clarification!

MartyG-RealSense, 2025-03-26:
1.  In the RealSense Viewer, Backlight Compensation is disabled by default in the **Controls** sub-section of the RGB options section.

2.  If by g-norm you mean the Y-Accel value, the ideal gravity target value is around 9.80 when the camera is stationary on a flat surface.  So if Y-Accel is 9.58 when the camera is flat and stationary then this would indicate a less than good calibration.
In the Viewer, when the Motion Module is enabled you can see the Gyro and Accel values by hovering the mouse cursor over the Gyro or Accel panel.

If you go to the RealSense Viewer's Motion Module IMU section and expand open the **Controls** sub-section, there should be an **Enable Motion Correction** option that can 'fix' raw IMU data and so bring Y-Accel closer to the 9.80 ideal.

[image]

In the Accel values in the Viewer, is Y-Accel still around 9.58 even with Enable Motion Correction enabled (indicated by a blue box beside the option).  If it is then it would be recommended to attempt a calibration of the IMU with the `rs-imu-calibration.py` tool.

lyc1140889141, 2025-03-26:
Hi, thanks for your response!

According to the **Intel RealSense Depth Cameras IMU Calibration Tool.pdf**, I tested the IMU in six positions and observed the following g-norm values:

- **Position 1** – Mounting screw pointing **down**, device facing out → **g-norm = 9.71-9.73**
- **Position 2** – Mounting screw pointing **left**, device facing out → **g-norm = 9.74-9.76**
- **Position 3** – Mounting screw pointing **up**, device facing out → **g-norm = 9.75**
- **Position 4** – Mounting screw pointing **right**, device facing out → **g-norm = 9.73**
- **Position 5** – Viewing direction **facing down** → **g-norm = 9.58-9.59**
- **Position 6** – Viewing direction **facing up** → **g-norm = 9.87-9.89**

I have a few follow-up questions:

1. When referring to **Y-Accel**, do you mean the g-norm observed in **Position 1** (where the white axis in RealSense Viewer is in the opposite direction to the green Y-axis) or in **Position 3** (where the white axis is aligned with the green Y-axis)?

[image]

[image]

2. Why do different positions produce different g-norm values?  
   Ideally, after performing **`rs-imu-calibration.py`**, should the g-norm in all six positions be consistently close to **9.80**?  
   If not, what level of variation is considered acceptable?  

   Additionally, due to the **physical shape of the D455, the USB port location, and the actual connection cables**, some positions are difficult to maintain in a perfectly vertical or aligned orientation.  
   Could this misalignment affect the accuracy of the `rs-imu-calibration.py` self-calibration process?

3. How does the `rs-imu-calibration.py` tool, which adjusts g-norm to ideal values as observed in RealSense Viewer, affect IMU calibration when directly recording the `/camera/imu` topic in a rosbag and later using IMU calibration toolboxes such as `imu_utils` or `allan_variance_ros`?  

   Specifically, is running `rs-imu-calibration.py` considered a **prerequisite step** for accurate IMU noise modeling, and how does it influence or relate to the results obtained from these calibration toolboxes?

Thanks for your insights!

MartyG-RealSense, 2025-03-26:
The camera will often be at Position 1 orientation (mounting screw down, facing directly forwards).  A Y gravity value of 9.71 to 9.73 is therefore good and so calibration of the IMU may not be needed.

In regard to your questions:

1.  When talking about Y-Accel I am assuming that the camera is in Position 1 - camera facing forward, mounting screw facing down.

2.  The g-values seem to be fairly consistent for most positions except 5 and 6.  I do not personally know why those orientations have a markedly different g-value.  However, if Position 1 has a Y-Accel value that is reasonably close to 9.80 (at least over 9.7) then it can be assumed that the calibration in general is likely to be good.

If there is difficulty in resting the camera in a particular orientation then I would recommend attaching to the mounting screw the miniature tripod supplied with the camera if you have it.  If you do not have the supplied tripod then the screw-thread of the hole has a standardized size that is compatible with a wide range of camera tripod products.

3.  If you are recording IMU data to a rosbag in ROS by publishing the `/imu` topic using the RealSense ROS wrapper then I recommend setting the ROS wrapper parameter **enable_motion_correction** to true so that the raw IMU data is 'fixed' in real-time to reduce inaccuracy, as described at https://github.com/IntelRealSense/realsense-ros/issues/3195

Typically, RealSense IMUs already have very good calibration when purchased and so do not require a re-calibration with `rs-imu-calibration.py`.  

I admittedly do not have detailed knowledge of IMU noise modelling.  In my opinion however this calibration tool is no longer as important as it used to be and the majority of RealSense users with an IMU-equipped camera should not need to use it.  RealSense 400 Series cameras changed to using an alternative IMU hardware component after mid 2022 (from Bosch BMI055 to the Bosch BMI085).

lyc1140889141, 2025-03-28:
Hi again, and thank you for your continued support!

I have a few follow-up questions regarding IMU calibration and configuration using the D455:

1. Based on my understanding of the IMU Calibration Tool whitepaper, the purpose of `rs-imu-calibration.py` is to correct systematic errors such as coordinate axis misalignment, scale factor inconsistency, and constant bias (zero offset).  
   After that, toolboxes like `imu_utils` or `allan_variance_ros` are used to model random errors, specifically white noise and bias random walk.  
   Is this interpretation correct?  
   In my case, I observed that Position 5 had a significantly lower g-norm and Position 6 had a significantly higher g-norm, which led me to run the calibration.

2. Considering that different sensor modules may interfere with each other and cause fluctuations in IMU data,  
   during standalone IMU calibration, is it recommended to **disable RGB-related launch arguments** such as `color_width`, `color_height`, and `enable_color` to minimize noise?  
   Or should I keep RGB enabled in order to **better simulate the real data collection conditions** (e.g., IMU + RGB enabled together as used in VINS-Mono)?

3. Does the D455 IMU exhibit a high sensitivity to **ambient temperature**?  
   Should I try to ensure that the **calibration temperature is similar to the actual operating environment** (e.g., outdoor conditions), in order to minimize temperature-induced bias drift?

4. Regarding the priority between RealSense Viewer settings and custom launch file parameters:  
   For options such as `Auto Exposure Priority`, `Global Time Enabled`, `Backlight Compensation` (under RGB Camera), and `Enable Motion Correction` (under the Motion Module),  
   if I enable any of these in the RealSense Viewer, it seems that the settings are **persistently saved** to the device and continue to take effect even when launching with a custom ROS launch file.

   For example, if I enable `Global Time Enabled` in the Viewer, and then launch the camera using a custom launch file **without setting any parameter related to global time**,  
   will the system still use global time for RGB and IMU streams due to the Viewer’s prior configuration?

   In other words, do RealSense Viewer settings get stored on the device and override or silently apply **even when not specified in the launch file**?

Thanks again for your time and detailed explanations!

MartyG-RealSense, 2025-03-28:
1.  Your interpretation is correct.

2.  I believe that the IMU calibration script will be using whatever the default values for accel and gyro are on your camera.  For example, before mid 2022 the default lower value for accel was 63.  After mid 2022 when the IMU component was replaced in RealSense cameras for a different but similar component (changing Bosch BMI055 to Bosch BMI085), 100 became the minimum supported accel value instead of 63.

If two **cfg** stream configuration instructions were inserted after line 120 to specifically configure the accel and gyro streams to 200 and 400 instead of using the defaults of 100 accel and 200 gyro, you could likely do that.  As far as I am aware the frequency does not affect the calibration process though.

https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-imu-calibration/rs-imu-calibration.py#L221

3.  When running `rs-imu-calibration.py`, ideally this script should be the only program running, and the ROS wrapper, Viewer, etc should not be open and active at the same time as the calibration script as they are not necessary for the IMU calibration process to function.  The RGB and depth sensors will not interfere with the IMU data, but running additional programs alongside `rs-imu-calibration.py` will be consuming the computer's processing resources needlessly.

4.  Very high temperature (where the camera's internal temperature is above 42 degrees C) can negatively affect the depth image, but not the IMU.  The only relationship between the IMU and depth is that each IMU data packet is timestamped using the depth sensor hardware clock to allow temporal synchronization between gyro, accel and depth frames.  There is no need to take account of temperature during IMU calibration.

5.  Ideally, the Viewer should be closed when a ROS wrapper session is active as the Viewer is not required for the ROS wrapper to function.  Some RealSense users in the past have found though that settings made in the Viewer when it and the wrapper are both open at the same time get applied to the ROS launch.  I do not know why this occurs, only that it can occur.

Global Time is enabled by default on RealSense 400 Series cameras and so does not require it to be enabled with a custom setting.  The ROS wrapper will therefore make use of Global Time by default.  It can be disabled in the ROS wrapper though if a user wishes as ROS has its own built-in time syncing mechanism.  In some rare cases - like in https://github.com/IntelRealSense/realsense-ros/issues/2905 - the data in a ROS session will perform better if Global Time is disabled.

lyc1140889141, 2025-03-29:
Thank you very much for your detailed and patient explanations throughout this discussion.  

I'll consider this issue resolved.  
Wishing you all the best and have a great day ahead!

MartyG-RealSense, 2025-03-29:
Thank you very much.  I'm pleased that I could help.  As you consider the issue resolved, I will close it.  Thanks again!

closed 2025-03-29 by MartyG-RealSense