realsenseai/realsense-ros #3323: Seemingly random and very high gyro values

DivAgarwal1, 2025-03-27:
Hello,

I have a D455 camera I'm running the wrapper with "ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true unite_imu_method:=2 pointcloud.enable:=true enable_color:=true enable_depth:=false" and then echoing the "/camera/imu" topic. The video attached below shows how the angular velocity values are far larger than expected and seemingly entirely random when the camera is left alone or just gently lifted up. Even without rotating or turning the camera, the values become extremely high, often exceeding 30-50 radians/s. Does anyone know why this is happening/how to fix it? Any help is appreciated!

https://drive.google.com/file/d/1Kre-QY3BsHZMT6Ou17k18OFje014fx2Y/view?usp=sharing

MartyG-RealSense, 2025-03-28:
Hi @DivAgarwal1  Are your IMU values more stable if you change `unite_imu_method` from 2 to 1 (Copy mode instead of Linear Interpolation mode).

DivAgarwal1, 2025-03-28:
Yes, I have tried setting unite_imu_method to 1 and 0 in addition to 2 and have seen the same behavior.

MartyG-RealSense, 2025-03-29:
Do you have the librealsense SDK's example programs installed on your computer?  If you do, are you able to run the **rs-motion** IMU example program?  If the IMU values fluctuate wildly in that program too then it could suggest that the issue is with the IMU or its calibration rather than the ROS2 wrapper.

You could also enable the **Motion Module** in the realsense-viewer tool if you have it to see how sensitive the IMU is in that tool.

The SDK has a Python IMU calibration program called **rs-imu-calibration.py**.  This calibration tool displays IMU values in real-time.  IT asks the user to position the camera on a flat surface and orient it at a particular angle, then let go of the camera.  If the printout of the angle value continues to fluctuate despite the camera being stationary and flat then it can suggest a problem with the IMU.

https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-imu-calibration

In cases where such fluctuations occur even when the camera is stationary, this prevents the camera's IMU from being calibrated with the Python script.  If applying 'fixing' correction of the raw data or enabling Copy mode does not stop the fluctuations then there is usually nothing that can be done to stop them, unfortunately.

Are you using librealsense 2.55.1?  If 2.55.1 is installed but the firmware version is older than 5.16.0.1 then it can result in corrupted accel and gyro values due to a variable gyro sensitivity feature introduced in 2.55.1 that requires firmware 5.16.0.1.

[image]

DivAgarwal1, 2025-03-29:
After using the realsense-viewer tool I find that the values are very reasonable and what I would expect. I am on librealsense2 v2.55.1 according to apt, but I also have ros-galactic-librealsense2 with version v2.51.1. When I run the launch script, this is the version that prints in the console output. The realsense-viewer also shows that I have firmware version 5.16.0.1. I don't think calibration is the issue since realsense-viewer has the correct data.

MartyG-RealSense, 2025-03-30:
There were not many RealSense ROS users who reported that they were using ROS2 Galactic.  My recollection is that users who were using Galactic with the RealSense ROS2 wrapper tended to experience problems more often than Foxy users.

If you are able to change ROS2 versions and cannot use a newer version such as Humble then it might be worth trying a downgrade to Foxy and reinstalling the 4.51.1 wrapper for it to see if your IMU stability improves.

MartyG-RealSense, 2025-04-10:
Hi @DivAgarwal1  Do you require further assistance with this case, please?  Thanks!

DivAgarwal1, 2025-04-12:
I have not solved the issue yet, but I'm currently putting this part of the project on hold. I'll likely end up revisiting it soon.

MartyG-RealSense, 2025-04-12:
Okay, thanks very much for the update.  I look forward to your next report in the near future.  I will keep this issue open in the meantime.  Good luck!

DivAgarwal1-UIUC, 2025-04-22:
Hey Marty,

Per your advice, I decided to try updating to using ROS2 Humble. I installed the ROS distro and the 'ros-humble-librealsense2' and 'ros-humble-realsense2' packages all from deb packages. 

I tried running 'ros2 launch realsense2_camera rs_launch.py enable_color:=false enable_infra:=false enable_infra1:=false enable_infra2:=false enable_depth:=false enable_gyro:=true enable_accel:=true unite_imu_method:=2' and was met with another error. Running without the gyro and accel streams works fine, but when those are enabled, setting up the node runs into issues. I have attached the output below.

[image]

MartyG-RealSense, 2025-04-23:
Are you using the camera on a PC or on an Nvidia Jetson computing board, please?  

If you are using a Jetson board, is it using JetPack 6?

closed 2025-05-11 by MartyG-RealSense