realsenseai/realsense-ros #3324: ros1-legacy wrapper does not work anymore?

Anniebhalla16, 2025-03-28:
Hi,
I read that ros1-lgeacy is not supported anymore, but saw recent commit to that branch dating Nov 2024.
'Not supported' means it does not work anymore, or it is not being further developed but is in working state?

I am trying to use ros1-legacy inside my ros-noetic dockerfile, however I run into this error.
Can anyone give me a clarity on this?

[image]

Thank you in advance :D

MartyG-RealSense, 2025-03-28:
Hi @Anniebhalla16  Development of the **ros1-legacy** wrapper was ceased after librealsense SDK version 2.50.0 and legacy wrapper version 2.3.2.   It has received a couple of very rare small updates after that time, but it should be assumed that the wrapper will no longer be updated (though we still provide support advice for it on this forum).

The 'failed to set power state' can indicate a failure to start the camera or a failure to detect it.

Can you confirm which RealSense camera model you are using, please?  Some more recently released RealSense camera models will not be supported in the legacy wrapper because development ended before those cameras were released.

If you are using ROS1 Noetic then Ubuntu **20.04** should be used.

If you are using a Docker container and launching it with a 'docker run' command, does it make a difference if you instead use **sudo docker run**

MartyG-RealSense, 2025-03-29:
1.  Is your Docker container installed on a PC computer or an Nvidia Jetson computing board, please?

2.  In realsense-viewer, is the depth stream able to be enabled and viewed?

3.  Can you confirm which librealsense version are you using?

GauravKumar9920, 2025-03-31:
Hii @MartyG-RealSense , I am facing the same error with intelsense d455,

I am running it in my pc with nvidia chip, inside distrobox, ubuntu 20.04 image
Yes, in the RealSense Viewer, the depth stream can be enabled and viewed.

MartyG-RealSense, 2025-04-01:
Providing the serial number in a RealSense ROS wrapper launch is only necessary when using more than one camera.

GauravKumar9920, 2025-04-03:
I see, In my case I am using realsense + livox mid-360.

MartyG-RealSense, 2025-04-03:
@GauravKumar9920  Thanks very much for the clarification.  I mean a serial number is needed when launching more than one **RealSense** camera.  

In a situation where there is a RealSense camera and a non-RealSense camera, it would be worth launching the RealSense camera and livox camera in two separate terminals if you are not doing so already.  Intel's guide for launching two RealSense cameras with `rs_camera.launch` on the same computer at the link below certainly recommends doing so.

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras#step-2-start-both-cameras

Does launching them in separate terminals (if you have not been doing so already) make a difference to the error that you are experiencing, please?

kokun-coffeebean, 2025-04-08:
i am trying to install D415 on Jetson Orin NX, and seeing almost same error as the screenshot,... is the problem solved?

GauravKumar9920, 2025-04-10:
> [@GauravKumar9920](https://github.com/GauravKumar9920) Thanks very much for the clarification. I mean a serial number is needed when launching more than one **RealSense** camera.

Apologies for late reply, was a bit occupied. 
I am just launching a single real-sense camera.

> Does launching them in separate terminals (if you have not been doing so already) make a difference to the error that you are experiencing, please?

I am running them on separate terminals/ the issue persists even when we run it by isolating it.

MartyG-RealSense, 2025-04-10:
I am not able to test the setups of RealSense users' computers on my own one, unfortunately.

Another user at https://github.com/IntelRealSense/realsense-ros/issues/2620 had problems with using the ROS2 wrapper with Ubuntu 20.04 and Distrobox.   They then used a VM instead.

GauravKumar9920, 2025-04-10:
Thanks @MartyG-RealSense, I tried running it in docker the issue persists, I might have a solution for this issue, I will put it here if I succeed.

jiteshnagar1789, 2025-04-10:
try setting these rules https://github.com/IntelRealSense/librealsense/blob/master/config/99-realsense-libusb.rules

MartyG-RealSense, 2025-05-01:
Hi @GauravKumar9920  Do you require further assistance with this case, please?  Thanks!

GauravKumar9920, 2025-05-01:
Hey @MartyG-RealSense ! Sorry I forgot to mention this earlier. I switched to a different setup - a laptop with Ubuntu. Before that, I was using Pop OS 22.04 with distrobox on top. It seems like the issue only occurs when we run it on Pop OS.

MartyG-RealSense, 2025-05-01:
It's no trouble at all.  Thanks very much for the update!

Are you still aiming to get the ROS1 wrapper working on Pop OS?  There are a few past cases at https://github.com/IntelRealSense/realsense-ros/issues/1663 and https://github.com/IntelRealSense/librealsense/issues/5212 and https://answers.ros.org/question/373535/ where others were having problems with using that OS with RealSense.

MartyG-RealSense, 2025-05-01:
Okay, thanks!  As you have a working system right now, I will close this issue.  Please feel free to re-open it or create a new issue if you return to the Pop OS problem in the future.

closed 2025-05-01 by MartyG-RealSense