realsenseai/realsense-ros #3328: ROS2 foxy issue about D435i met hwmon command failed error

kight71, 2025-04-07:
tan@tan-Predator-PHN16-71:~$ ros2 run realsense2_camera realsense2_camera_node
[INFO] [1743991507.767826698] [camera]: RealSense ROS v4.51.1
[INFO] [1743991507.767960089] [camera]: Built with LibRealSense v2.51.1
[INFO] [1743991507.767974911] [camera]: Running with LibRealSense v2.51.1
[INFO] [1743991507.783146342] [camera]: Device with serial number 233522079395 was found.

[INFO] [1743991507.783208983] [camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:1a.0/0000:02:00.0/0000:03:02.0/0000:38:00.0/usb4/4-1/4-1:1.0/video4linux/video8 was found.
[INFO] [1743991507.783222179] [camera]: Device with name Intel RealSense D435I was found.
[INFO] [1743991507.783520639] [camera]: Device with port number 4-1 was found.
[INFO] [1743991507.783535762] [camera]: Device USB type: 3.2
[INFO] [1743991507.785670381] [camera]: getParameters...
[INFO] [1743991507.786312876] [camera]: JSON file is not provided
[INFO] [1743991507.786357305] [camera]: Device Name: Intel RealSense D435I
[INFO] [1743991507.786369490] [camera]: Device Serial No: 233522079395
[INFO] [1743991507.786379407] [camera]: Device physical port: /sys/devices/pci0000:00/0000:00:1a.0/0000:02:00.0/0000:03:02.0/0000:38:00.0/usb4/4-1/4-1:1.0/video4linux/video8
[INFO] [1743991507.786390342] [camera]: Device FW version: 05.16.00.01
[INFO] [1743991507.786400148] [camera]: Device Product ID: 0x0B3A
[INFO] [1743991507.786408952] [camera]: Sync Mode: Off
[ERROR] [1743991507.797524643] [camera]: An error has occurred while calling sensor for: depth_module.auto_exposure_limit:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ERROR] [1743991507.797872438] [camera]: An error has occurred while calling sensor for: depth_module.auto_gain_limit:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ERROR] [1743991507.798192724] [camera]: An error has occurred while calling sensor for: depth_module.auto_exposure_limit_toggle:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ERROR] [1743991507.798498998] [camera]: An error has occurred while calling sensor for: depth_module.auto_gain_limit_toggle:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[INFO] [1743991507.811197325] [camera]: Stopping Sensor: Stereo Module
[INFO] [1743991507.863833251] [camera]: Starting Sensor: Stereo Module
[INFO] [1743991507.902334601] [camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1743991507.902429772] [camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1743991507.902451631] [camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[INFO] [1743991507.902611492] [camera]: Stopping Sensor: RGB Camera
[INFO] [1743991507.904280354] [camera]: Starting Sensor: RGB Camera
[INFO] [1743991507.910001191] [camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[INFO] [1743991507.910076310] [camera]: Stopping Sensor: Motion Module
 07/04 10:05:08,007 WARNING [139754922297088] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[INFO] [1743991508.009581161] [camera]: Starting Sensor: Motion Module
[INFO] [1743991508.013080338] [camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 100
[INFO] [1743991508.013132589] [camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[INFO] [1743991508.013988290] [camera]: RealSense Node Is Up!
[WARN] [1743991508.017254344] [camera]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 07/04 10:05:08,017 WARNING [139754427356928] (sensor.cpp:1058) HID timestamp not found, switching to Host timestamps.

ROS version:foxy
SDK version: RealSense ROS v4.51.1
LibrealSense version: LibRealSense v2.51.1
I checked the page about the recommend version of LibrealSense, I found that this is the version recommended,and I have no idea about how to solve that error 
[ERROR] [1743991507.798498998] [camera]: An error has occurred while calling sensor for: depth_module.auto_gain_limit_toggle:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
may you help me? Does this error influence effect of D435i?

MartyG-RealSense, 2025-04-07:
Hi @kight71  The **auto_gain_limit_toggle** error can occur when a non-recommended camera firmware driver version is being used with a particular librealsense SDK version.  You are using firmware 5.16.0.1 but the recommended firmware for librealsense 2.51.1 is **5.13.0.50**

This feature is available in librealsense 2.55.1, whose recommended firmware is 5.16.0.1.  Using this firmware with older librealsense versions, such as 2.51.1, can result in the error because those older versions do not recognise the feature.  It should disappear if the firmware is downgraded to 5.13.0.50.

kight71, 2025-04-11:
Thank you for your help. I solve this error by using the method you mentioned. Really appreciate for your help!

closed 2025-04-11 by kight71

MartyG-RealSense, 2025-04-11:
You are very welcome.  I'm pleased to hear that a firmware downgrade to 5.13.0.50 resolved the problem.  Thanks very much for the update!