realsenseai/realsense-ros #3330: RealSense D455 Issues with ROS 2 Humble on Jetson

oblivione, 2025-04-11:
I’m using the Intel RealSense D455 camera with ROS 2 on a Jetson device. I initially set it up with JetPack 6.x (Ubuntu 22.04) and ROS 2 Humble, but ran into permission issues. The camera only worked with sudo, and the ROS 2 node wouldn’t run properly because ROS 2 is not meant to be launched with elevated permissions.
To resolve this, I downgraded JetPack 5.x (Ubuntu 20.04) with ROS 2 Foxy. Now everything works as expected—the RealSense camera works without sudo, and the ROS 2 camera node detects and streams from the camera correctly.

How to solve this issue for the ROS2 Humble?

Thank you

MartyG-RealSense, 2025-04-11:
Hi @oblivione  My research indicates that Humble can be used with JetPack 5, and also on older Ubuntu versions than 22.04, if the desktop version of Humble - **ros-humble-desktop** - is built from source code.  

The link below has an example of a guide for building Humble from source via ros-humble-desktop.

https://www.forecr.io/blogs/installation/how-to-install-ros-2-humble-hawksbill

oblivione, 2025-04-12:
@MartyG-RealSense , we tried the solution, but it did not work. I feel it's something to do with UVC permissions.

MartyG-RealSense, 2025-05-01:
Case closed due to no further comments received.

closed 2025-05-01 by MartyG-RealSense

xkns, 2025-05-21:
I've run into the exact same problem. I've installed librealsense with the libuvc backend as described above. Is there any fix other than downgrading the OS to JetPack 5? I couldn't find a related issue on the librealsense repo.

Interestingly the camera works fine when using `sudo`, but won't work when running as `root` or the normal user account.

MartyG-RealSense, 2025-05-21:
@xkns   If the libuvc backend installation of librealsense does not resolve the problem on its own then for some RealSense users, taking the further step after the libuvc installation of downgrading the camera's firmware driver to **5.13.0.50** fixes it.

This solution can only be used for older camera models such as D415, D435, D435i and D455 though.  

More recent camera models (D435f, D435if, D455f, D456, D457)  should not be downgraded to 5.13.0.50 as it will cause them to stop functioning.

xkns, 2025-05-21:
Thank you for your incredibly fast response!

Its a D455C. I've downgraded the firmware to 5.13.0.50. This has resolved an issue where the motion module wouldn't stream data, but the initial permission(?) issue persists.

`rs-enumerate-devices` was able to detect the camera running as the normal user just once after a reboot, but I haven't been able to reproduce that state yet. (Tried different combinations of reloading udev rules, rebooting and plugging into different USB ports)

MartyG-RealSense, 2025-05-21:
Yes, if sudo works but non-sudo does not then I would usually recommend reloading the udev rules.

If you can access the realsense-viewer tool, when it launches is there a pop-up error message in the top corner of the realsense-viewer window that says 'Multiple udev rules detected' ?

xkns, 2025-05-21:
realsense-viewer doesn't show any warnings. Running as normal user the log window is all green, and when running as with sudo there's a single error regarding a failed file download which seems unrelated.

xkns, 2025-05-21:
Yes I built up that exact command from the install script.

If this is starting to get too hard/much, I can also downgrade the OS to JetPack 5. That seems to have worked for the original poster.

xkns, 2025-05-21:
The MIPI driver appears to require additional hardware. I'll attempt downgrading to JetPack 5 first.

Thank you very much for you time and expertise :)