realsenseai/realsense-ros #3331: Feature Request - Support TF Prefixing

pondersome, 2025-04-12:
| Required Info	| |
| --- | --- |
| Camera Model	| D455 |
| Firmware Version | 5.16.0.1 |
| Operating System & Version	| Linux (Ubuntu 24.04) |
| Platform	| N100 x64 PC 16GB | 
| Librealsense SDK Version	| v2.55.1 |
| Language	| python |
| Segment	| Robot |
| ROS Distro	| humble |
| RealSense ROS Wrapper Version	| v4.55.1 |

This is a feature request to enhance support of TF prefix handling in the Realsense ROS2 Wrapper. The purpose is to enable a flexible organized hierarchy of camera related TF frame names. This should support single cameras as well as multiple cameras in the same robot network/domain.

When I launch with camera_namespace and camera_name arguments, the nodes and topics are properly namespaced and camera isolated. But the node-generated tf frames do not support the namespace. Instead they exist in the TF list named like:

```
camera1_link
camera1_color_frame
camera1_color_optical_frame
camera1_depth_frame	
camera1_depth_optical_frame

```

This at least allows us to isolate camera1 frames from camera2's, but it does not well-support more complex configurations. 

In my case, I have two cooperative robots, each with a 4 dof arm and D455s mounted at the end effector. I need to properly isolate tf frames from both cameras so I can attach them to the correct arm. A practice we follow is to mirror the naming of the tf frames off of the namespace hierarchy of the robot and its subsystems. For example we'll have:
```
Two chassis nodes in namespace /mybot1 and /mybot2
Two arm subsystems in namespace /mybot1/arm1 and /mybot2/arm1
```
Similarly we'd like the associated tf labelled with the same hierarchy including using a forward slash as the delimiter:
```
/mybot1/arm1/camera1_link
/mybot2/arm1/camera1_link
```
Even though we can walk the tree in a properly jointed hierarchy, it's also good to support just listing all of the frames by name and they will naturally be ordered and grouped in a similar fashion as the connected tree. This helps with organizing and navigating a complicated TF frame space.

While I use slashes to demark the organization of my frame names, users should have full freedom over the construction of the prefix, so my actual request is to honor a tf_prefix parameter like was apparently supported in the original ROS1 wrapper. This parameter should allow any characters that are legal in a TF frame name. To maintain compatibility with existing usage, the camera_name parameter can still be appended to an optionally provided prefix.

Note that I investigated accomplishing this with just the camera_name parameter, but since it must resolve to a legal node name, it can't support all of the legal TF frame name characters.

My initial hope was that the base_frame_id parameter could play a part in the solution, but the documentation on that parameter in the readme is spare and open to interpretation. More importantly I couldn't test it because it had no visible effect. It was missing as a supported parameter in the launch file and the launch file didn't warn that an unrecognized parameter was being used. I now understand that this is a way to replace "link" with a user supplied suffix string for the camera's base frame. And I see that the missing parameter has been added to the launch file in the ros2-development branch. It would also be nice to get a warning from the launch file if a non-supported parameter was supplied.

So I'm asking that TF frames have names constructed like:
```
[tf_prefix][camera_name]_[base_frame_id] # camera base frame only, or
[tf_prefix][camera_name]_[hard_coded_stream_name]
```

I see references touching on this need in some other issues:

https://github.com/IntelRealSense/realsense-ros/issues/3242

https://github.com/IntelRealSense/realsense-ros/issues/3119

MartyG-RealSense, 2025-04-13:
Hi @pondersome  Thanks so much for your Pull Request at https://github.com/IntelRealSense/realsense-ros/pull/3332 - I have put an **Enhancement** label on this case to signify that it should be kept open whilst your PR is active.

MartyG-RealSense, 2026-10-31:
Case closed due to the PR https://github.com/realsenseai/realsense-ros/pull/3332 linked to this issue having been merged in May 2025.

closed 2026-10-31 by MartyG-RealSense