realsenseai/realsense-ros #3334: enable_accel not working

jiteshnagar1789, 2025-04-15:
i am using d435if with ubuntu 20 and ros noetic . when i run rs_camera.launch with enable_accel:=true it regularly give control_transfer_return error and no message are published on topic camera/accel/sample .

MartyG-RealSense, 2025-04-15:
Hi @jiteshnagar1789  Which method did you use to install the librealsense SDK before you installed the ROS1 wrapper, please?

Also, are you using the official 1 meter long USB cable supplied with the camera, or another USB cable of your own choice?

MartyG-RealSense, 2025-04-15:
Thanks very much.  And which method did you then use to install the ROS1 wrapper?

If you built librealsense first then the wrapper has to be installed from source code using 'Method 2' of the wrapper instructions.

If 'Method 1' was used - `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` - then this installs both librealsense and the ROS1 wrapper at the same time, meaning that if you have already first installed librealsense then you can end up with two versions of librealsense on the same computer, which could lead to conflicts between them.

If you need to use realsense-viewer then it is necessary to install librealsense first and then the ROS1 wrapper second with Method 2, as the sudo apt-get instruction of Method 1 does not include realsense-viewer in the librealsense installation.

jiteshnagar1789, 2025-04-15:
i uninstalled all things and again installed all package are working now .

i want to make ir emitter off how i do that and then close the issue .

MartyG-RealSense, 2025-04-15:
I'm pleased to hear that it is working successfully after a re-installation!

You can enable the emitter on-off mode by adding `emitter_on_off:=true` to yur roslaunch instruction.

closed 2025-04-15 by jiteshnagar1789