realsenseai/realsense-ros #3335: Ubuntu 24.04: multi-camera problems with D435f

denseishin, 2025-04-16:
Hello everyone,

I'm running ROS 2 Jammy on Ubuntu 24.04 with the recent kernel. 
I installed librealsense from source and applied the kernel patches. Version: v2.56.3
realsense-ros version: v4.56.3

When I try to start my own launchfile for 2 D435f's, I get the following errors:
```
[realsense2_camera_node-2] [WARN] [1744822777.517596245] [camera2.camera2]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.74482e+12,Warn,Frames Timeout
[realsense2_camera_node-1] [WARN] [1744822777.813211803] [camera2.camera1]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.74482e+12,Warn,Frames Timeout

```
In that time, only RGB images are correctly getting published and no other images despite the topics being listed.

On the other hand, if I run rs_multi_camera_launch.py without any additional parameters, only one camera effectively outputs images (both RGB and depth), while the other one seems broken. 

In both cases, the affected camera seems to change everytime I restart the nodes.

How can I get my cameras to run on my current system?

Error logs for rs_multi_camera_launch.py:
```[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [25764]
[INFO] [realsense2_camera_node-2]: process started with pid [25765]
[INFO] [static_transform_publisher-3]: process started with pid [25766]
[static_transform_publisher-3] [WARN] [1744821819.867029610] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-3] [INFO] [1744821819.877671478] [static_transform_publisher_oU8mM7e8YhH8tofW]: Spinning until stopped - publishing transform
[static_transform_publisher-3] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-3] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-3] from 'camera1_link' to 'camera2_link'
[realsense2_camera_node-1] [0m[INFO] [1744821820.188807428] [camera1.camera1]: RealSense ROS v4.56.3[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.188824973] [camera2.camera2]: RealSense ROS v4.56.3[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.189018161] [camera2.camera2]: Built with LibRealSense v2.56.3[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.189057865] [camera2.camera2]: Running with LibRealSense v2.56.3[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.189028154] [camera1.camera1]: Built with LibRealSense v2.56.3[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.189087546] [camera1.camera1]: Running with LibRealSense v2.56.3[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.202136907] [camera1.camera1]: Device with serial number 244222074383 was found.
[realsense2_camera_node-1] [0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.202154489] [camera1.camera1]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video2 was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.202162541] [camera1.camera1]: Device with name Intel RealSense D435 was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.202280990] [camera1.camera1]: Device with port number 2-1 was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.202289679] [camera1.camera1]: Device USB type: 3.2[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.202323424] [camera1.camera1]: getParameters...[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.202357497] [camera2.camera2]: Device with serial number 244222074383 was found.
[realsense2_camera_node-2] [0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.202377857] [camera2.camera2]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video2 was found.[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.202385019] [camera2.camera2]: Device with name Intel RealSense D435 was found.[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.202477100] [camera2.camera2]: Device with port number 2-1 was found.[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.202485380] [camera2.camera2]: Device USB type: 3.2[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.202509893] [camera2.camera2]: getParameters...[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.202623551] [camera1.camera1]: JSON file is not provided[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.202638151] [camera1.camera1]: Device Name: Intel RealSense D435[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.202644154] [camera1.camera1]: Device Serial No: 244222074383[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.202651603] [camera1.camera1]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video2[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.202661138] [camera1.camera1]: Device FW version: 5.13.0.55[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.202668681] [camera1.camera1]: Device Product ID: 0x0B07[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.202676144] [camera1.camera1]: Sync Mode: Off[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.202807971] [camera2.camera2]: JSON file is not provided[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.202820970] [camera2.camera2]: Device Name: Intel RealSense D435[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.202826893] [camera2.camera2]: Device Serial No: 244222074383[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.202831485] [camera2.camera2]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video2[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.202836655] [camera2.camera2]: Device FW version: 5.13.0.55[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.202841076] [camera2.camera2]: Device Product ID: 0x0B07[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.202844963] [camera2.camera2]: Sync Mode: Off[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.286068542] [camera1.camera1]: Set ROS param depth_module.depth_profile to default: 848x480x30[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.286160799] [camera2.camera2]: Set ROS param depth_module.depth_profile to default: 848x480x30[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.286333253] [camera1.camera1]: Set ROS param depth_module.infra_profile to default: 848x480x30[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.286387038] [camera2.camera2]: Set ROS param depth_module.infra_profile to default: 848x480x30[0m
[realsense2_camera_node-2] [0m[INFO] [1744821820.290122158] [camera2.camera2]: Set ROS param rgb_camera.color_profile to default: 640x480x30[0m
[realsense2_camera_node-1] [0m[INFO] [1744821820.290335675] [camera1.camera1]: Set ROS param rgb_camera.color_profile to default: 640x480x30[0m
[... 99 lines cut ...]
[realsense2_camera_node-2] [0m[INFO] [1744821821.847787556] [camera2.camera2]: Starting Sensor: Depth Module[0m
[realsense2_camera_node-2]  16/04 18:43:41,848 ERROR [127233964377792] (librealsense-exception.h:52) xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy
[realsense2_camera_node-2]  16/04 18:43:41,848 ERROR [127233964377792] (rs.cpp:256) [rs2_open_multiple( sensor:0x64a0f2fc4500, profiles:0x73b7b4035e80, count:1 ) Backend] xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy
[realsense2_camera_node-2] [31m[ERROR] [1744821821.848101639] [camera2.camera2]: /home/agrobot/workspaces/cibv2/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:494:An exception has been thrown: xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy[0m
[realsense2_camera_node-2] [31m[ERROR] [1744821821.848131952] [camera2.camera2]: /home/agrobot/workspaces/cibv2/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:376:An exception has been thrown: xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy[0m
[realsense2_camera_node-2] Failed to start device: xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy
[realsense2_camera_node-2] [0m[INFO] [1744821821.848172383] [camera2.camera2]: Stopping Sensor: Depth Module[0m
[realsense2_camera_node-2] [0m[INFO] [1744821821.848293095] [camera2.camera2]: Stopping Sensor: RGB Camera[0m
[realsense2_camera_node-1] [0m[INFO] [1744821821.848616604] [camera1.camera1]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30[0m
[... cut ...]
[realsense2_camera_node-1]  16/04 18:43:42,414 ERROR [136707510896320] (global_timestamp_reader.cpp:253) Error during time_diff_keeper polling: acquire: Cannot open '/dev/video2 Last Error: No such file or directory
[realsense2_camera_node-1]  16/04 18:43:42,414 ERROR [136707320055488] (backend-v4l2.cpp:3060) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 52 error: No such device
[realsense2_camera_node-1]  16/04 18:43:42,414 ERROR [136707330541248] (backend-v4l2.cpp:3060) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 44 error: No such device
[last two lines on repeat again...]
```

MartyG-RealSense, 2025-04-17:
Hi @denseishin  Which recent kernel version are you using, please?  The librealsense SDK 2.56.3 supports patching for kernels up to kernel 6.8 on Jammy.

[image]

When there are two cameras attached, if multiple camera detection is not working then the first camera detected will be used.  This is why the camera that is detected and used can change when the nodes are restarted.

Do the cameras each publish their topics correctly if you open two terminals and launch each camera in its own terminal with its own defined name?

**Terminal 1**
`ros2 launch realsense2_camera rs_launch.py camera_namespace:=camera1 camera_name:=D435F_1`

**Terminal 2**
`ros2 launch realsense2_camera rs_launch.py camera_namespace:=camera2 camera_name:=D435F_2`

MartyG-RealSense, 2025-04-17:
What computer are you using the cameras with, please?  Some low-end computing boards, such as Raspberry Pi, will be only capable of running a single RealSense camera.

denseishin, 2025-04-17:
At the moment, I'm using a Dell Precision 3440.

MartyG-RealSense, 2025-04-17:
If one or the other camera is detected (but not both), it suggests that both cameras are capable of being in working order, just only one of them at a time.

Yes, I was also going to suggest testing the `usb_port_id` parameter to identify cameras by their port IDs (which the log indicates are 2-1 and 2-6).

Does the situation improve if you unplug the micro-sized end of the USB cable from the side of the D435f cameras, turn the connector around the other way and re-insert them into the camera.  USB-C cables are two-way insertion at the micro-sized end, and cameras can sometimes perform better with the micro-connector in one particular orientation of the two available.

Have you tried resetting the cameras by adding `initial_reset:=true` to the launch command?

denseishin, 2025-04-17:
Turning the connector around doesn't change anything. setting `initial_reset` to true in my own launchfile seemingly doesnt improve the situation. The reset is taking place though for both cameras. With USB port IDs specified, rs_multi_camera_launch.py seems to work as intended

MartyG-RealSense, 2025-04-17:
I'm pleased to hear that  rs_multi_camera_launch.py is working correctly when using port IDs.

MartyG-RealSense, 2025-04-17:
The depth stream does not require the infrared topics to be enabled in order to be published, so if you do not need the IR topics then disabling them will definitely free up bandwidth.

Reducing stream resolution, FPS or both will reduce bandwidth if there is a time when you need the infrared streams.

denseishin, 2025-04-17:
Thank you for your help :)

closed 2025-04-17 by denseishin