realsenseai/realsense-ros #3339: Unable get frame in rtabmap with Intel Realsense D457.

tirthankarhalder, 2025-04-25:
I am setting up rtabmap with Inter Realsense D457.  The camera is starting but frames are not showing in rtabmapviz.

Os distribution:
`Linux tirtha-Predator-PH315-51 6.11.0-24-generic #24~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Mar 25 20:14:34 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`

Ros Distribution : ROS Jazzy

Other dependencies : 
`LibRealSense v2.55.1`
`RealSense ROS v4.55.1`

D457 Firmware : 

when I run `ros2 launch rtabmap_examples realsense_d400.launch.py`

I am getting three warning:
`[rtabmap_viz-4] [WARN] [1745588034.653848243] [rtabmap_viz]: rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ ros2 topic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Ajusting topic_queue_size (10) and sync_queue_size (10) can also help for better synchronization if framerates and/or delays are different. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[rtabmap_viz-4] rtabmap_viz subscribed to (exact sync):
[rtabmap_viz-4]    /odom \
[rtabmap_viz-4]    /camera/color/image_raw \
[rtabmap_viz-4]    /camera/aligned_depth_to_color/image_raw \
[rtabmap_viz-4]    /camera/color/camera_info \
[rtabmap_viz-4]    /odom_info
`

and 

`[rgbd_odometry-2] [WARN] [1745588037.473438857] [rgbd_odometry]: rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[rgbd_odometry-2] rgbd_odometry subscribed to (exact sync, topic_queue_size=10, sync_queue_size=5):
[rgbd_odometry-2]    /camera/color/image_raw,
[rgbd_odometry-2]    /camera/aligned_depth_to_color/image_raw,
[rgbd_odometry-2]    /camera/color/camera_info
`

and 

`[rtabmap-3] [WARN] [1745588037.633895165] [rtabmap]: rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ ros2 topic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Ajusting topic_queue_size (10) and sync_queue_size (10) can also help for better synchronization if framerates and/or delays are different. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[rtabmap-3] rtabmap subscribed to (exact sync):
[rtabmap-3]    /odom \
[rtabmap-3]    /camera/color/image_raw \
[rtabmap-3]    /camera/aligned_depth_to_color/image_raw \
[rtabmap-3]    /camera/color/camera_info \
[rtabmap-3]    /odom_info`

Initialization Output : 
`[INFO] [launch]: All log files can be found below /home/tirtha/.ros/log/2025-04-25-18-57-31-976229-tirtha-Predator-PH315-51-100923
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [100926]
[INFO] [rgbd_odometry-2]: process started with pid [100927]
[INFO] [rtabmap-3]: process started with pid [100928]
[INFO] [rtabmap_viz-4]: process started with pid [100929]
[realsense2_camera_node-1] [INFO] [1745587652.357389279] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1745587652.360351086] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1745587652.360581161] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1745587652.409331094] [camera.camera]: Device with serial number 234322307380 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1745587652.409438003] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/video4linux/video2 was found.
[realsense2_camera_node-1] [INFO] [1745587652.409481310] [camera.camera]: Device with name Intel RealSense D455 was found.
[realsense2_camera_node-1] [INFO] [1745587652.409915433] [camera.camera]: Device with port number 1-1 was found.
[realsense2_camera_node-1] [INFO] [1745587652.409949348] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1745587652.409968862] [camera.camera]: Device 234322307380 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1745587652.410072188] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1745587652.410545094] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1745587652.410588181] [camera.camera]: Device Name: Intel RealSense D455
[realsense2_camera_node-1] [INFO] [1745587652.410607362] [camera.camera]: Device Serial No: 234322307380
[realsense2_camera_node-1] [INFO] [1745587652.410621801] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/video4linux/video2
[realsense2_camera_node-1] [INFO] [1745587652.410638544] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1745587652.410650371] [camera.camera]: Device Product ID: 0x0B5C
[realsense2_camera_node-1] [INFO] [1745587652.410667374] [camera.camera]: Sync Mode: Off
[rgbd_odometry-2] [INFO] [1745587652.445227440] [rgbd_odometry]: Odometry: frame_id               = camera_link
[... 44 lines cut ...]
[rtabmap-3] [INFO] [1745587652.474945245] [rtabmap]: rtabmap(maps): octomap_tree_depth         = 16
[rtabmap-3] [INFO] [1745587652.504233477] [rtabmap]: rtabmap: frame_id      = camera_link
[rtabmap-3] [INFO] [1745587652.504267140] [rtabmap]: rtabmap: map_frame_id  = map
[rtabmap-3] [INFO] [1745587652.504279988] [rtabmap]: rtabmap: log_to_rosout_level  = 4
[rtabmap-3] [INFO] [1745587652.504288238] [rtabmap]: rtabmap: initial_pose  = 
[rtabmap-3] [INFO] [1745587652.504295665] [rtabmap]: rtabmap: use_action_for_goal  = false
[rtabmap-3] [INFO] [1745587652.504304728] [rtabmap]: rtabmap: tf_delay      = 0.050000
[rtabmap-3] [INFO] [1745587652.504318408] [rtabmap]: rtabmap: tf_tolerance  = 0.100000
[rtabmap-3] [INFO] [1745587652.504332328] [rtabmap]: rtabmap: odom_sensor_sync   = false
[rtabmap-3] [INFO] [1745587652.504340271] [rtabmap]: rtabmap: pub_loc_pose_only_when_localizing = false
[rtabmap-3] [INFO] [1745587652.504347621] [rtabmap]: rtabmap: wait_for_transform = 0.200000
[rtabmap-3] [INFO] [1745587652.504355549] [rtabmap]: rtabmap: gen_scan  = false
[rtabmap-3] [INFO] [1745587652.504362727] [rtabmap]: rtabmap: gen_depth  = false
[realsense2_camera_node-1] [INFO] [1745587652.523628527] [camera.camera]: Set ROS param depth_module.depth_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1745587652.524366511] [camera.camera]: Set ROS param depth_module.infra_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1745587652.529093842] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1745587652.529917902] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1745587652.529993842] [camera.camera]: Set ROS param accel_fps to default: 100
fps to default: 100
[realsense2_camera_node-1] [INFO] [1745587652.535854703] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1745587652.535993130] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1745587652.543735638] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1745587652.547026594] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15`

I unable to solve this. Kindly guide me to solve this.

MartyG-RealSense, 2025-04-25:
Hi @tirthankarhalder  Do you get the same results if you instead use the launch instruction below, please?

`ros2 launch rtabmap_ros realsense_d400.launch.py`

It is worth highlighting that Jazzy support for 24.04 was not introduced until the next RealSense ROS wrapper version **4.56.1**, which should be used with librealsense beta version 2.56.1.

[image]

tirthankarhalder, 2025-04-25:
Thank you so much for the reply. @MartyG-RealSense 
After running `ros2 launch rtabmap_ros realsense_d400.launch.py`
I am getting 
`file 'realsense_d400.launch.py' was not found in the share directory of package 'rtabmap_ros'`

Should I use the same realsense_d400.launch.py from rtabmap_examples?

MartyG-RealSense, 2025-04-26:
My knowledge of rtabmap is admittedly limited, so thanks very much for your patience.

I note that your USB cable connection is being detected in the log as slow **USB 2.1**.  If you unplug the micro-sized USB connector from the base of the D457 camera, turn the connector around the other way and re-insert it (the cable is two-way insertion at the micro-sized end) and perform the launch, does the launch then detect the connection type as **3.2** ?   You can find the detected USB speed on the **Device USB type** line of the log.

If you try again `ros2 launch rtabmap_examples realsense_d400.launch.py` - towards the end of your launch log, does it contain anywhere the line **RealSense Node is Up!** as this is the usual indicator that the launch has completed without exiting and the RealSense topics have been published.

If you achieve a USB 3.2 connection but the log does not have **RealSense Node is Up!** near the end then yes, I would certainly consider building the beta versions as your next step.

MartyG-RealSense, 2025-04-27:
Are you attempting to install librealsense using colcon build, please?  If you are, colcon is only used when building the ROS wrapper from source code.  It should not be used when building librealsense from source code.  

Running the edited libuvc_installation.sh script file in the 'root' directory location  of the librealsense source code folder should be sufficient for installation of librealsense to occur.

tirthankarhalder, 2025-04-29:
Thank you so much, @MartyG-RealSense. I built the realsense-ros from the source code. The versions are matching now as you mentioned earlier. But the same RTABMap error is coming. I have also raised an issue in their forum.

MartyG-RealSense, 2025-04-29:
Have you performed an ordinary non-RTABMAP launch of the RealSense ROS wrapper and then used **ros2 topic list** to confirm that the RealSense topics can be published correctly, please?

`ros2 launch realsense2_camera rs_launch.py`

MartyG-RealSense, 2025-05-14:
That's excellent news, @tirthankarhalder - thanks very much for the update about your success!

closed 2025-05-24 by MartyG-RealSense