realsenseai/realsense-ros #3341: D455 for RGB recording

lyc1140889141, 2025-04-29:
Hi RealSense team, it's me again.

I'm using the RealSense D455 with the `ros1-legacy` branch of the RealSense ROS wrapper.  
Currently, I'm configuring the D455 for **outdoor RGB + external IMU recording** for use in **visual-inertial SLAM**, and I have a few questions regarding best practices and recommended settings.

---

### 1. Reducing RGB Recording Size

I'm recording RGB frames by subscribing to `/d455/color/image_raw` at 640x480, 30Hz.  
However, the rosbag grows quickly — about **1.58GB per minute**, which is quite large for long-term recording.

I noticed from `rs-enumerate-devices` that **Y8 (grayscale)** format is supported and from RealSense Viewer it is easy to switch from `RGB8` to `Y8`.  

[image]
[image]

Is there a way to switch to Y8 **within the ROS launch file**, or by adjusting something in `base_realsense_node.cpp`?

---

### 2. Compressed Topics vs Raw RGB

I also found alternative topics such as `/d455/color/image_raw/compressed` and `/d455/color/image_raw/theora`.

[image]

How do these **compressed formats** affect:
- image quality
- data delay (latency)

compared to `/d455/color/image_raw`?

Also, what exactly is the difference between the **`compressed`** and **`theora`** versions?

---

### 3. Timestamp Source Without Hardware Sync 

I’m no longer using the built-in D455 IMU and instead rely on an external one.

Since I'm only recording RGB from the D455, and there is no hardware sync trigger, when publishing `/d455/color/image_raw`, is the `header.stamp` generated by calling `ros::Time::now()`?

[image]

---

Thanks a lot for your time and support!

MartyG-RealSense, 2025-04-29:
Hi @lyc1140889141  Thanks very much for your questions.

1.  If you wish to use Y8 infrared instead of RGB8 RGB then you can add the parameters below to your ROS1 roslaunch command:

`enable_color:=false enable_infra1:=true`

2.  A RealSense user at https://github.com/IntelRealSense/realsense-ros/issues/3105#issuecomment-2146157266 published tables comparing the megabytes per second (MB/s) data size of different compressed topics, including `camera/color/image_raw/` and `camera/color/image_raw/compressed` versus `color/image_raw/theora`.  Their tests were performed in ROS2 but should give you an idea of what to expect in ROS1.

Theora is a different video compression format that is suited to video compression.

3.  A D455 will have camera Global Time enabled by default.  But ROS also has its own timing mechanism and so you can disable the RealSense global time mechanism if you wish and use ROS's timing by adding `global_time_enabled:=false` to your roslaunch command.

More information about global time versus ROS time in regard to the ROS1 wrapper can be found at https://github.com/IntelRealSense/realsense-ros/issues/796

lyc1140889141, 2025-04-29:
Hi,

I'm using D455 only for recording the single RGB camera.

### 1. Y8 Output from RGB Camera

Is it possible to directly output the RGB camera in **Y8 grayscale** format at roslaunch time?  
If not, I would consider subscribing to `/d455/color/image_raw/compressed` instead.

---

### 2. How to set launch arguments

(1) Should options like `enable_color:=false`, `enable_infra1:=true`, or `global_time_enabled:=false` be set by adding them **to the roslaunch command line**, or by adding **new arguments inside the `.launch` file**?

(2) It seems that no matter whether `global_time_enabled` is disabled, when I open RealSense Viewer after starting roslaunch, the RGB camera always shows `Global Time Enabled` as checked by default.  
How can I correctly verify if `global_time_enabled` is really disabled when running ROS?

---

### 3. rqt_reconfigure effect on `/color/image_raw/compressed`

When I change `jpeg_quality` settings under `/d455/color/image_raw/compressed` in `rqt_reconfigure`, does this immediately take effect for the compressed images being published?

---

Thanks again for your help!

MartyG-RealSense, 2025-04-30:
1.  In the ROS2 wrapper, it is possible to select a different stream format in the launch.  Please try adding the parameter below to your ROS2 launch command.

`rgb_camera.color_format:=Y8`

2.  Some users find that the status of `global_time_enabled` does not change when it is set in the launch instruction.  I don't think you'd be able to change it via launch file in ROS2 so that it was set to false without having to set it yourself manually unless you edited the rs_launch.py launch file to add global_time_enabled to its list of supported parameters.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/realsense2_camera/launch/rs_launch.py#L24

I believe that your understanding of global_time_enabled = true compared to when it is false is correct.

3.  rqt_reconfigure settings are applied dynamically during runtime, so any parameter changed with this method should be updated immediately.

lyc1140889141, 2025-05-01:
Hi,

Following up on this issue as I'm still using the D455 only for RGB recording.

### 1. Outdoor Magenta Tint from RGB Camera

(1) As mentioned in [librealsense issue #7155](https://github.com/IntelRealSense/librealsense/issues/7155), the D455 RGB camera shows a strong magenta tint in bright outdoor lighting when auto exposure is enabled.  
Is this a known hardware limitation of the RGB sensor, or could it be resolved through firmware or parameter tuning?

(2) I tried disabling auto exposure and setting `saturation=0` as a workaround. This seems to remove the magenta distortion visually.  
In this case, can the image be considered grayscale for downstream usage (e.g., feature tracking with VINS-Mono)?  
Does `/d455/color/image_raw` or `/d455/color/image_raw/compressed` now output grayscale-like images, or is further conversion (e.g., to mono8) still recommended?

---

### 2. Persistent Image Noise  

(1) Regardless of lighting condition (indoor or outdoor), resolution (640x480 or 1280x720), or frame rate (30fps or 60fps), I observe visible noise or "dancing pixel artifacts" in the image when viewed via `rqt_image_view`.  
Is this noise a result of sensor hardware (e.g., CMOS limitations), or is it expected under the current RealSense ROS wrapper/software pipeline?

(2) How significant is this kind of noise for VSLAM applications (e.g., KLT optical flow or feature tracking)?  
Would you consider this noise to resemble Gaussian white noise, and are there recommended manual adjustments in `realsense-viewer` under the RGB Camera options (e.g., exposure, gain, etc.) that can help reduce this visual noise?

---

Thank you again for your support!

MartyG-RealSense, 2025-05-01:
1.  The RGB purple / pink tint that can occur when outdoors is a characteristic of the D455 hardware and not a bug.  As advised at https://github.com/IntelRealSense/librealsense/issues/7155#issuecomment-691356656 it is typically caused by a combination of strong light and a very short exposure time.

Applying a physical ND filter or Linear Polarizer filter product over the lenses on the front of the camera or defining a **Region of Interest (ROI)** on the lower half of the RGB image can help to negate the tinting.

Thin-film linear polarizing filters can be obtained inexpensively on stores such as Amazon by searching for the term **linear polarizing filter sheet**

In the RealSense ROS2 wrapper you can change the stream format, such as changing RGB color's default full-color RGB8 format to monochrome Y8.  As you are using the ROS1 wrapper though, that feature is not available and so Y8 infrared would be the only way to obtain a monochrome RGB-like image.

2.  You might experience noise on the image indoors if the camera is used in a location with overhead fluorescent ceiling strip lights, as these contain heated gases that flicker at frequencies that are difficult to see with the human eye.  Configuring an RGB setting called **Power Line Frequency** in the RGB controls of realsense-viewer can help to reduce interference from such lights.  For example, if you are in a European country then setting it to '50' may help, whilst North American regions are suited to the '60' setting.

You could try manual RGB exposure value '78' and FPS value '5' in realsense-viewer to see whether the noise reduces, as these settings are good for minimizing RGB image lag / blur.

In regard to the effects of noise on SLAM or identifying noise types, that is outside of my knowledge, unfortunately.  I do apologize.

lyc1140889141, 2025-05-01:
Hi,

Thanks again. I’d like to clarify two related points regarding the use of the infrared camera as a grayscale image source.

### 1. Infrared Camera Output Format  
When launching with `enable_infra1:=true`, the topic `/d455/infra1/image_raw` appears with a cyan-tinted image in `rqt_image_view`.  
Unlike the RGB camera, the Stereo Module in `realsense-viewer` only exposes `exposure` and `gain` controls, without options like `saturation`, `hue`, or `color format`.  
Is there a way — via `roslaunch`, `rqt_reconfigure`, or `realsense-viewer` — to force the infrared camera to publish **true grayscale (Y8 / mono8)** images instead of the colored tint?  
If not, what causes the cyan appearance, and is there any way to correct it in the current RealSense ROS1 framework?

### 2. Infrared Y8 vs. RGB8 with saturation=0  
The RGB camera allows manually setting `saturation=0`, which visually removes color, but it still outputs RGB8.  
In terms of accuracy, consistency, and noise performance, would you recommend using the native infrared stream (Y8) over RGB8 with `saturation=0` to obtain grayscale image data?

Thanks!

MartyG-RealSense, 2025-05-02:
1.  The Y8 infrared image should always be monochrome.  I have not seen a previous case where it has been tinted with color.  However, RealSense users typically view the images in RViz rather than rqt_image_view, so I do not know if this is an issue that only occurs in rqt_image_view.  Are you able to display Y8 infrared in RViz to see if the tinting also occurs there, please?

The changing of RGB parameters such as hue and saturation is supported in the ros1-legacy wrapper.  You can test it during runtime after launch by inputting into the terminal the command `rosrun rqt_reconfigure rqt_reconfigure` to open the rqt_reconfigure graphical configuration interface, and then select **RGB Camera** from its side-panel to view the adjustable RGB options.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy?tab=readme-ov-file#set-camera-controls-using-dynamic-reconfigure-params

Each parameter that can be dynamically changed in rqt_reconfigure also has an equivalent parameter of the same name that can be defined in the launch file.

2.  Visually I do not have a preference between monochrome infrared Y8 over monochrome infrared RGB.  A key benefit of Infrared though is that it can make use of the invisible light projected by the camera's IR emitter component to brighten the image when the lighting level in a location is dim.  As RGB cannot take advantage of the projector's light, its image will be darker in low-light environments.

lyc1140889141, 2025-05-03:
Thanks for your helpful reply about the Y8 infrared image, that made things clearer!

I have another question related to intrinsics calibration using Kalibr in a ROS environment.

When I calibrate the D455 RGB camera using Kalibr and get the intrinsics (K and D matrices), I’m trying to understand which coordinate frame is being used for the camera. Is it using the ROS FLU convention (`camera_color_frame`), or the OpenCV-style RDF convention (`camera_color_optical_frame`)?

I found issue [#743](https://github.com/IntelRealSense/realsense-ros/issues/743) that explains the difference between these two frames. Since the meaning of the intrinsics depends on the camera’s coordinate system, I’d like to know if the RealSense camera and wrapper follow OpenCV’s RDF convention when publishing images.

This matters because it affects how I understand and check whether the calibrated T_ic (the transform between IMU and RGB camera) matches the actual hardware setup, using RViz or other 3D visualization tools.

[image]
[image]
[image]

Thanks again!

MartyG-RealSense, 2025-05-03:
The RealSense coordinate system is compatible with the OpenCV pinhole camera model,

A key difference between the RealSense coordinate system and the ROS standard one is that the ROS standard is X-Forward, Y-Left, Z-Up, whilst with the RealSense axes, the positive x-axis points to the right, the positive y-axis points down and the positive z-axis points forward

Where there are differences between the RealSense coordinate system and the ROS standard, the RealSense ROS wrapper automatically converts RealSense axes to the ROS standard axes.

If Kalibr is used for calibration then this automatic conversion of RealSense axes to ROS standard would not take place as Kalibr is not designed specifically with RealSense in mind.

lyc1140889141, 2025-05-06:
Thanks, Marty — that helps a lot. Appreciate your support!

MartyG-RealSense, 2025-05-06:
You are very welcome.  I'm pleased that I could help!

lyc1140889141, 2025-05-06:
Sorry if I'm being overly detailed — and I may have misunderstood part of your explanation.

I'm still trying to fully understand the coordinate frames involved in extrinsic calibration. I believe the coordinate system that each data stream (camera and IMU) uses during calibration needs to be known by Kalibr, or at least can be inferred from the bag.

(1) For the external IMU, I confirmed with the vendor that it uses the standard ROS FLU frame.

(2) For the RGB camera, when I run `rostopic echo /d455/color/camera_info`, `/image_raw/header`, or `/image_raw/compressed/header`, the `frame_id` is always `d455_color_optical_frame`. And according to the ROS wiki, both `sensor_msgs/CameraInfo` and `sensor_msgs/Image` define this frame as having +x to the right, +y down, and +z forward — which matches the RDF convention.

So is it reasonable to conclude that the image data published on `/image_raw` and `/image_raw/compressed` is expressed in the RDF coordinate system?

Thanks again for your time and support!

lyc1140889141, 2025-05-07:
Thanks again for your help.

Furthermore, after performing extrinsic calibration between the D455's internal RGB camera and IMU using Kalibr, and visualizing the result in 3D, I noticed that both `/d455/color/image_raw` and `/d455/imu` use the frames `camera_color_optical_frame` and `camera_imu_optical_frame` respectively. This was also confirmed by checking the `frame_id` field in the header of each topic via `rostopic echo`.

These frames also appear in RViz with RDF orientation, and the calibration result seems to align with the physical layout.

[image]
[image]

Given this, can I confirm that when I subscribe directly to `/d455/color/image_raw`, the image data itself is already in OpenCV's RDF coordinate system, without any further conversion? That is, are the image pixels published in a frame that matches the standard OpenCV pinhole camera model?

Really appreciate the clarification!

MartyG-RealSense, 2025-05-07:
The raw frames of streams published in the RealSense ROS wrapper will have the same 'positive x-axis right, the positive y-axis down, positive z-axis points forward' "RDF" coordinate system that frames in the librealsense SDK have (the coordinate system that is compatible with the OpenCV pinhole camera model).  The ROS wrapper is essentially a layer running on top of librealsense.

lyc1140889141, 2025-05-07:
That explains it — thanks a lot for your help!  
I think the issue is completely resolved now.  
Best regards!

closed 2025-05-17 by MartyG-RealSense