realsenseai/realsense-ros #3342: D435 ROS topics publishing but not sending data

Jesseblair, 2025-04-29:
Hey @MartyG-RealSense as a frequent visitor of the issues page for the ROS wrapper I've come to the conclusion that you're the man with all the answers. All of the issues I have had so far I've been able to find solutions to however I've come across a real doozy.

I need to pull data from the pointcloud and raw image topics to use in an ROS package I am working on. I had this working just fine for quite a while on a USB 2.0 cable, I picked up a 3.0 cable so I could utilize the full potential of the camera and for whatever reason cannot get data to stream to the published topics. I am not sure if changing the cable has anything to do with my issue but as far as I'm aware that is the only factor that changed when things broke so its worth mentioning. It should be noted that when I swapped back to a 2.0 cable I still have issues. 

I am able to view everything properly in the realsense viewer which I assume means its not a hardware issue. The two topics I need are /camera/camera/color/image_raw and camera/camera/depth/color/points. For whatever reason when I run `ros2 topic echo /camera/camera/color/image_raw` it either prints nothing at all or just one "frame" of data and stops. Similarly when running `ros2 topic echo camera/camera/depth/color/points` I never get any output and at the same time the terminal I launched the wrapper from throws:
`[realsense2_camera_node-1] [WARN] [1745958420.330525264] [camera.camera]: No stream match for pointcloud chosen texture Process - Color`

I have uninstalled and reinstalled librealsense2 and the ROS wrapper hoping that would fix my issue but luck was not on my side.

I have tried a plethora of different launch arguments including but not limited to enable_sync, initial_reset, enable_color, align_depth.enable, pointcloud.allow_no_texture_points and even tried setting the depth and color profiles to 640x480x30, although that seemed to cause more issues. 

Any help would be appreciated, I look forward to hearing from you.

Launch args that worked with 2.0 cable:
`ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true pointcloud.stream_filter:=2 pointcloud.stream_index_filter:=0`

Terminal output after launch:
```shell
[INFO] [launch]: All log files can be found below /home/jesse/.ros/log/2025-04-29-16-24-04-558819-jetson-11315
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [11316]
[realsense2_camera_node-1] [INFO] [1745958244.979081088] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1745958244.979479278] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1745958244.979526960] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1745958245.021675691] [camera.camera]: Device with serial number 250122077816 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1745958245.021891731] [camera.camera]: Device with physical ID /sys/devices/platform/bus@0/3610000.usb/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1745958245.021927892] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1745958245.022333411] [camera.camera]: Device with port number 2-2 was found.
[realsense2_camera_node-1] [INFO] [1745958245.022374628] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1745958245.022589676] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1745958245.023136127] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1745958245.023178721] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1745958245.023208258] [camera.camera]: Device Serial No: 250122077816
[realsense2_camera_node-1] [INFO] [1745958245.023232355] [camera.camera]: Device physical port: /sys/devices/platform/bus@0/3610000.usb/usb2/2-2/2-2:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1745958245.023255748] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1745958245.023278436] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1745958245.023299205] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1745958245.205911301] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1745958245.207017613] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1745958245.213589783] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1745958245.225299575] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 640x480x30
[realsense2_camera_node-1] [INFO] [1745958245.244005745] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1745958245.244659112] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1745958245.401076836] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1745958245.413000780] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1745958245.443228832] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1745958245.453594448] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1745958245.457823815] [camera.camera]: RealSense Node Is Up!
```

MartyG-RealSense, 2025-04-30:
Hi @Jesseblair  The message `No stream match for pointcloud chosen texture Process - Color`  indicates that RGB color frames are not being reliably received, perhaps because frames are dropping.  If echoing the color topic shows that there is no data then this could explain why that message is occurring.

Does the pointcloud generate successfully if you disable the color stream and enable the Infra1 infrared stream and use the infrared image to texture the pointcloud instead by setting the filter to '1' instead of '2' ?

`ros2 launch realsense2_camera rs_launch.py enable_color:=false enable_infra1:=true pointcloud.enable:=true pointcloud.stream_filter:=1`

Jesseblair, 2025-04-30:
Hey @MartyG-RealSense, thanks for getting back to me. Using the infrared image to texture the pointcloud gives me the same result, no data stream. I have noticed that if I monitor the pointcloud topic with echo and restart the ROS wrapper the topic will publish a few (1-3) "frames" of data then stop.

MartyG-RealSense, 2025-04-30:
Does performance improve if you unplug the micro-sized end of the USB cable from the side of the D435 camera, turn the connector around the other way and re-insert it (the cable is two-way insertion at the micro-sized end).

Jesseblair, 2025-04-30:
No luck, still getting the no stream match warning and occasionally it publishes one frame then hangs

Jesseblair, 2025-04-30:
That worked!

Is there a way I can change the launch arguments with a config file using ros2 run?

I appreciate your help.

MartyG-RealSense, 2025-04-30:
You could try creating a **yaml** configuration file containing the launch file parameters and then load in the yaml file with a command within the ros2 run instruction.

A guide and YouTube video for doing so can be found at the link below.

https://roboticsbackend.com/ros2-yaml-params/

MartyG-RealSense, 2025-05-11:
Hi @Jesseblair  Do you require further assistance with this case, please?  Thanks!

Jesseblair, 2025-05-15:
Hey @MartyG-RealSense I had everything running fine for a while but I'm now running into similar issues but this time on the image_raw topic. Using ros2 run to launch works fine for the point cloud data but now the rgb image data is only publishing very briefly after the node launch then hanging just like the point cloud data was previously.

Jesseblair, 2025-05-15:
To add onto this, when I limit the color, depth, and infra feeds to 15fps they seem to all work without issue. When running at 30+ the above issues appear.

MartyG-RealSense, 2025-05-16:
At the start of this case you bought a USB 3.0 cable.  If you are currently using the 2.0 cable again, have you tried the 3.0 one with the config that works at 15 FPS?

MartyG-RealSense, 2025-05-22:
Are you able to use 30 FPS if you reduce the value of the **jpeg_quality** parameter for depth, infrared and color - for example, changing it from '95' to '70'.