realsenseai/realsense-ros #3343: Are there any parameters in the launch file that allows me to adjust the exposure of the color camera?

YiKang-Liu, 2025-04-30:
I use ros in ros noetic to run realsense D455 camera，the ros package version is 2.3.2
Due to my usage needs, I will sometimes use the camera outdoors. However, when used outdoors, light problems can cause the output image of a color camera to be white and nothing can be seen.

I tried to follow the online method and added this parameter in the launch file
  <arg name=“color_autoexposure” default=“false”/>
  <arg name=“color_exposure” default=“100”/>
  <arg name=“color_gain” default=“64”/>

      <arg name=“color_autoexposure” value=“$(arg color_autoexposure)”/> <arg name=“color_autoexposure” default=“$(arg color_autoexposure)”/>
      <arg name=“color_exposure” value=“$(arg color_exposure)”/> <arg name=“color_autoexposure” default=“$(arg color_autoexposure)”/>
      <arg name=“color_gain” value=“$(arg color_gain)”/>

But it's not working.After launching the launch file, the published image remains blank. I would like to know if it is possible to set the color camera lens to manual exposure in the launch file and adjust its parameters
The full launch file I'm using right now is as follows.

<launch>

  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>
  
  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>
  
  <arg name="depth_width"         default="848"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>
  
  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>
  
  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>
  
  <arg name="color_width"         default="848"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>
  
  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="15"/>
  <arg name="infra_fps"           default="15"/>
  <arg name="color_fps"           default="15"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>
  
  <arg name="enable_pointcloud"         default="true"/>
  
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_ANY"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>
  
  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>
[... 69 lines cut ...]
      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="reconnect_timeout"        value="$(arg reconnect_timeout)"/>
      <arg name="wait_for_device_timeout"  value="$(arg wait_for_device_timeout)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="stereo_module/exposure/1" value="$(arg stereo_module/exposure/1)"/>
      <arg name="stereo_module/gain/1"     value="$(arg stereo_module/gain/1)"/>
      <arg name="stereo_module/exposure/2" value="$(arg stereo_module/exposure/2)"/>
      <arg name="stereo_module/gain/2"     value="$(arg stereo_module/gain/2)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
      <arg name="ordered_pc"               value="$(arg ordered_pc)"/>
      
      <arg name="color_autoexposure"       value="$(arg color_autoexposure)"/>
      <arg name="color_exposure"           value="$(arg color_exposure)"/>
      <arg name="color_gain"               value="$(arg color_gain)"/>
    </include>
  </group>
</launch>

Hopefully one of you kind souls can help me with this question

MartyG-RealSense, 2025-04-30:
Hi @YiKang-Liu  You should be able to test your settings in ROS1 during runtime after launch if you input the commands below into the terminal.  The changes will update in real-time whilst the color stream is running.

```
rosrun dynamic_reconfigure dynparam set /camera_gap/rgb_camera/ enable_auto_exposure 0
rosrun dynamic_reconfigure dynparam set /camera/rgb_camera/ exposure 100
rosrun dynamic_reconfigure dynparam set /camera/rgb_camera/ gain 64
```

In the ROS1 launch file, you can configure RGB settings by adding a **rosparam** code block to the file.

```
<rosparam>
 /camera/rgb_camera/enable_auto_exposure: false
 /camera/rgb_camera/exposure: 100
 /camera/rgb_camera/gain: 64
</rosparam>
```

YiKang-Liu, 2025-05-04:
The complete launch file and the nodelet.xml I'm using right now is as follows

```xml
rs_camera.launch
<launch>

  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>
  
  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>
  
  <arg name="depth_width"         default="848"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>
  
  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>
  
  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>
  
  <arg name="color_width"         default="848"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>
  
  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="15"/>
  <arg name="infra_fps"           default="15"/>
  <arg name="color_fps"           default="15"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>
  
  <arg name="enable_pointcloud"         default="true"/>
  <!-- <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/> -->
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_ANY"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>
  
  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>
  
  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>
  
  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="4.0"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="true"/>
  <arg name="reconnect_timeout"         default="6.0"/>
  <arg name="wait_for_device_timeout"   default="-1.0"/>
  <arg name="unite_imu_method"          default="linear_interpolation"/>
[... 314 lines cut ...]
        params = {
            'enable_auto_exposure': False,
            'exposure': 100,
            'gain': 32
        }
        
        rospy.loginfo("Setting camera parameters: %s", str(params))
        result = client.update_configuration(params)
        rospy.loginfo("Camera parameters set successfully: %s", str(result))
        
        rospy.sleep(2)
        
    except Exception as e:
        rospy.logerr("Failed to set camera parameters: %s", str(e))
        raise  

if __name__ == '__main__':
    try:
        set_camera_params()
    except rospy.ROSInterruptException:
        pass
    except Exception as e:
        rospy.logerr("Camera parameter setting failed: %s", str(e))
```
Is there something wrong with my launch or nodelet file?

MartyG-RealSense, 2025-05-04:
Please add the parameters to your launch file as a `<rosparam> </rosparam>` block of code as advised above and place the block of code immediately before the list of `<param name>` parameters referred to in the code block.

For example:

```
<rosparam>
 /camera/rgb_camera/enable_auto_exposure: false
 /camera/rgb_camera/exposure: 100
 /camera/rgb_camera/gain: 64
</rosparam>

<param name="rgb_camera/enable_auto_exposure" type="bool" value="$(arg enable_auto_exposure)"/>
<param name="rgb_camera/exposure" type="int" value="$(arg color_exposure)"/>
<param name="rgb_camera/gain" type="int" value="$(arg color_gain)"/>
```

closed 2025-05-17 by MartyG-RealSense