realsenseai/realsense-ros #3344: Incorrect frame IDs in RGBD message

jbarry-bdai, 2025-04-30:
The RGBD camera message reports that the color image and depth image are in different frames.  However, according to the REAMDE, the depth and color images in this message should be aligned, which means that the frame IDs should be the same.  Additionally, the overall frame is hard-coded to be "camera_rgbd_optical_frame" which both doesn't exist and doesn't properly use the name parameter.

MartyG-RealSense, 2025-05-01:
Hi @jbarry-bdai  The RGBD topic is not widely used.  I would recommend instead using the RealSense ROS2 wrapper's `/camera/camera/aligned_depth_to_color/image_raw` aligned depth-color topic in combination with time-tag sync instead of the RGBD topic.

`ros2 launch realsense2_camera rs_launch.py enable_sync:=true align_depth.enable:=true enable_color:=true enable_depth:=true` 

Under the new format for frame_id where `camera_name` is used instead of `<stream>_optical_frame_id`, the default name for the camera if you do not define a name yourself will be **camera**

Are you able to use the frame_id `camera_color_optical_frame`

jbarry-bdai, 2025-05-02:
Hi @MartyG-RealSense !  Thanks.  We do give the camera a custom name.

On the RGBD topic, which frame are the depth and color images in?  They do appear to be aligned.  A small amount of testing led us to believe it was actually the depth optical frame although the color optical frame would make more sense if I understand the implementation correctly.

MartyG-RealSense, 2025-05-03:
A RealSense ROS user at https://github.com/IntelRealSense/realsense-ros/issues/3070 who echoed the RGBD topic with `ros2 topic echo /camera/camera/rgbd` received the output that its frame_id was separately **camera_color_optical_frame** for color and **camera_depth_optical_frame** for depth.  

It likely has two separate frame ids because the RGBD topic publishes RGB and depth together in the same message.

jbarry-bdai, 2025-05-07:
@MartyG-RealSense yes but I think the frame for the depth image or color image is _wrong_ because they have been (according to the documentation) converted to be in the _same_ frame.  There has been post processing done but it looks like the header wasn't updated to reflect that.

MartyG-RealSense, 2025-05-07:
Are you referring to the line below from the RGBD topic documentation, please?

**enable_sync: let librealsense sync between frames, and get the frameset with color and depth images combined**

It describes how depth and color are combined into a **frameset**.  A frameset is a set of frames that were captured at approximately the same time (like a box with a collection of frames within it).  

You can have frames from multiple types of streams within a frameset (such as depth and color) if more than one stream type is enabled, but the frames remain separate even though they are stored in the same frameset 'container'.  

My recommendation is still to use the `/camera/camera/aligned_depth_to_color/image_raw` aligned depth-color topic instead of the RGBD topic if it is possible for you to do so, as there is very little information available about how the RGBD topic functions other than the official description.

MartyG-RealSense, 2025-06-08:
Hi @jbarry-bdai  I commented two weeks ago but you may not have seen it if you are having problems with email notifications.  Do you require further assistance with this case, please?  Thanks!

closed 2025-07-27 by MartyG-RealSense