realsenseai/realsense-ros #3346: uvc streamer watchdog triggered on endpoint

elecDeath, 2025-05-04:
Trying to use the ros wrapper for d435i but keep getting uvc streamer watchdog triggered on endpoint: 130 error. Installed using sudo apt-get install ros-noetic-realsense2-camera.
Command - roslaunch realsense2_camera rs_camera.launch initial_reset:=true

[INFO] [1746328615.987303193]: Initializing nodelet with 12 worker threads.
[INFO] [1746328616.093846405]: RealSense ROS v2.3.2
[INFO] [1746328616.093890357]: Built with LibRealSense v2.50.0
[INFO] [1746328616.093949540]: Running with LibRealSense v2.50.0
[INFO] [1746328616.128371331]:  
[INFO] [1746328616.311759117]: Device with serial number 215322070919 was found.

[INFO] [1746328616.311805960]: Device with physical ID 2-2-2 was found.
[INFO] [1746328616.311813417]: Device with name Intel RealSense D435I was found.
[INFO] [1746328616.312195335]: Device with port number 2-2 was found.
[INFO] [1746328616.312241000]: Device USB type: 3.2
[INFO] [1746328616.312269705]: Resetting device...
[INFO] [1746328622.405152886]:  
[INFO] [1746328622.603482673]: Device with serial number 215322070919 was found.

[INFO] [1746328622.603516463]: Device with physical ID 2-2-3 was found.
[INFO] [1746328622.603530356]: Device with name Intel RealSense D435I was found.
[INFO] [1746328622.603910325]: Device with port number 2-2 was found.
[INFO] [1746328622.603932492]: Device USB type: 3.2
[INFO] [1746328622.605024747]: getParameters...
[INFO] [1746328622.632450812]: setupDevice...
[INFO] [1746328622.632479142]: JSON file is not provided
[INFO] [1746328622.632490194]: ROS Node Namespace: camera
[INFO] [1746328622.632499800]: Device Name: Intel RealSense D435I
[INFO] [1746328622.632536657]: Device Serial No: 215322070919
[INFO] [1746328622.632543477]: Device physical port: 2-2-3
[INFO] [1746328622.632550474]: Device FW version: 05.13.00.50
[INFO] [1746328622.632559748]: Device Product ID: 0x0B3A
[INFO] [1746328622.632566602]: Enable PointCloud: Off
[INFO] [1746328622.632573076]: Align Depth: Off
[INFO] [1746328622.632580220]: Sync Mode: Off
[INFO] [1746328622.632617293]: Device Sensors: 
[INFO] [1746328622.724133075]: Stereo Module was found.
[INFO] [1746328622.739502581]: RGB Camera was found.
[INFO] [1746328622.739677833]: Motion Module was found.
[INFO] [1746328622.739721759]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[INFO] [1746328622.739763860]: num_filters: 0
[INFO] [1746328622.739778390]: Setting Dynamic reconfig parameters.
[INFO] [1746328626.011746598]: Done Setting Dynamic reconfig parameters.
[INFO] [1746328626.012145413]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[INFO] [1746328626.013172555]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[INFO] [1746328626.013202549]: setupPublishers...
[INFO] [1746328626.015136505]: Expected frequency for depth = 30.00000
[INFO] [1746328626.041336314]: Expected frequency for color = 30.00000
[INFO] [1746328626.058368038]: setupStreams...
 04/05 08:47:06,293 WARNING [140055117031168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[INFO] [1746328626.343386044]: SELECTED BASE:Depth, 0
[INFO] [1746328626.351125219]: RealSense Node Is Up!
[WARN] [1746328626.411028146]: 
 04/05 08:47:06,411 WARNING [140055117031168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 04/05 08:47:06,461 WARNING [140055117031168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 04/05 08:47:06,512 WARNING [140055117031168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 04/05 08:47:06,866 WARNING [140055117031168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 04/05 08:47:06,921 WARNING [140055117031168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 04/05 08:47:06,972 WARNING [140055117031168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 04/05 08:47:07,022 WARNING [140055117031168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 04/05 08:47:07,073 WARNING [140055117031168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 04/05 08:47:07,125 WARNING [140055117031168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 04/05 08:47:07,518 ERROR [140055072929536] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 04/05 08:47:07,679 WARNING [140055117031168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 04/05 08:47:08,290 ERROR [140054756468480] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 04/05 08:47:09,033 WARNING [140055117031168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 04/05 08:47:09,622 ERROR [140054756468480] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 04/05 08:47:10,085 WARNING [140055117031168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 04/05 08:47:10,849 ERROR [140055072929536] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 04/05 08:47:12,182 ERROR [140055072929536] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 04/05 08:47:12,288 ERROR [140054756468480] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 04/05 08:47:13,514 ERROR [140055072929536] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 04/05 08:47:13,954 ERROR [140054756468480] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 04/05 08:47:15,513 ERROR [140055072929536] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 04/05 08:47:16,286 ERROR [140054756468480] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 04/05 08:47:17,142 WARNING [140055117031168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 04/05 08:47:17,179 ERROR [140055072929536] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 04/05 08:47:17,618 ERROR [140054756468480] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

@MartyG-RealSense

elecDeath, 2025-05-04:
Running it on my laptop, camera works fine with realsense viewer.but gave the aforementioned error when running using sudo apt-get install ros-noetic-realsense2-camera on a clean ubuntu install.

elecDeath, 2025-05-04:
1.i installed using method 1, got that error 
2. reinstalled ubuntu, tried method 2 and checked realsense viewer, worked fine
3. reinstalled ubuntu, back to method 1, got that error again

MartyG-RealSense, 2025-05-04:
How does the ROS launch behave if you remove the `initial_reset:=true` instruction from your roslaunch command?  Do the **control_transfer returned** warnings and endpoint errors still occur?

Are the published topics able to be viewed successfully in RViz despite the messages?

elecDeath, 2025-05-04:
yes they still occur, i can view the topics but the fps is too low, rostopic hz says it is around 10 and it keeps dropping

MartyG-RealSense, 2025-05-04:
Thank you very much for the information.  It's good to hear that the topics are functioning, though at a slow rate.

Let's try an **RGBD** type of launch next by following the steps below.

1.  Install support for an RGBD launch by inputting the command below into the terminal.

**sudo apt-get install ros-noetic-rgbd-launch**

2.  Run the **rs_rgbd.launch** launch file instead of rs_camera.launch

`roslaunch realsense2_camera rs_rgbd.launch`

elecDeath, 2025-05-05:
still getting the same error...

MartyG-RealSense, 2025-05-05:
Does FPS improve or the number of warning / error messages reduce if you set the stream resolution and FPS to very low values using the launch command below?

`roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=6 color_width:=640 color_height:=480 color_fps:=6`

elecDeath, 2025-05-06:
yes, the uvc error is gone now and the fps remains at 6 fps, i am fine with turning off color and depth and just want the infra streams at 30 fps for visual odomtery, so i tried the following command -
roslaunch realsense2_camera rs_camera.launch initial_reset:=true enable_color:=false enable_depth:=false enable_infra1:=true enable_infra2:=true infra_fps:=30

but this still causes the watchdog triggered error

elecDeath, 2025-05-06:
i need stereo images, i tried 6, 15, 30 fps, but only 6 seems to work, i need it to be 30 or 15 atleast

elecDeath, 2025-05-06:
i tried 424x240@30 it seems to work fine, but not 640x480@15 or 640x480@30, how can i get that working?

elecDeath, 2025-05-06:
i tried running it on 640x480@30, it worked once, but now again it has started giving me the error

MartyG-RealSense, 2025-05-06:
Are you using the official 1 meter USB cable that is supplied with the D435i camera or another USB cable of your own choice, please?

When you are launching the ROS wrapper, is the realsense-viewer tool closed (as it is not required in order to use the ROS wrapper)

Does stability improve if you unplug the micro-sized end of the USB cable from the side of the camera, turn the connector around the other way and re-insert it?  A USB-C cable (full size connector on one end, micro-sized connector on the other) is two-way insertion at the micro-sized end.

elecDeath, 2025-05-06:
yes using the official cable, i dont have realsense viewer as i am currently using method 1, about flipping the cable, do you mean flipping the usb c cable the other way, i tried that it doesnt affect anything. like my current cable is usb a to usb c, i flipped the usb c side

MartyG-RealSense, 2025-05-07:
Yes, I meant flipping the USB cable the other way.  Thanks for letting me know that it did not work for you.

When you reinstalled the ROS wrapper from source code using method 2 and realsense-viewer worked fine, did you still have the same error in the ROS wrapper?

Would it be possible for you to use **compressed** ROS topics, which may be able to achieve the 30 FPS that you are aiming for?  Compressed topics can be published by installing the **compressed_image_transport** package for Noetic with the command below.

`ros-noetic-compressed-image-transport`

Compressed topics for Infra1 and Infra2 should then be published:

```
/camera/infra1/image_rect_raw/compressed
/camera/infra2/image_rect_raw/compressed
```

elecDeath, 2025-05-08:
yes i still had the same issue when using method 2, the compressed topics are laggy as well :(

MartyG-RealSense, 2025-05-08:
During streaming after launch, does FPS improve if you open the **rqt_reconfigure** graphical configuration interface with the command `rosrun rqt_reconfigure rqt_reconfigure` and then click on the 'rgb_camera' category option in the side-panel and de-select (untick) the RGB option 'auto_exposure_priority' to disable it?

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy?tab=readme-ov-file#set-camera-controls-using-dynamic-reconfigure-params

[image]

elecDeath, 2025-05-09:
doesnt seem to reduce the uvc errors unfortunately

MartyG-RealSense, 2025-05-09:
And there is still lag if you launch with **rosrun** instead of roslaunch?

`rosrun realsense2_camera rs_camera.launch`

elecDeath, 2025-05-09:
laptop@laptop:~$ rosrun realsense2_camera rs_camera.launch
[rosrun] Couldn't find executable named rs_camera.launch below /opt/ros/noetic/share/realsense2_camera
[rosrun] Found the following, but they're either not files,
[rosrun] or not executable:
[rosrun]   /opt/ros/noetic/share/realsense2_camera/launch/rs_camera.launch

i get the following error

MartyG-RealSense, 2025-05-10:
There is not much else that can be tried, unfortunately.

1.  Does it make a difference if you launch with either of the launch file commands below?

`roslaunch realsense2_camera opensource_tracking.launch enable_color:=false enable_depth:=false enable_infra1:=true enable_infra2:=true infra_fps:=30`

`roslaunch realsense2_camera demo_pointcloud.launch enable_color:=false enable_depth:=false enable_infra1:=true enable_infra2:=true infra_fps:=30`

2.  You stated earlier in this discussion that 424x240 at 30 FPS worked.  If the most important thing is that both infra streams are enabled and the FPS is 30, would you be able to use 424x480 with visual odometry?

3.  In situations where a performance problem has been unsolveable, a couple of RealSense ROS users have attained the performance level that they need by writing their own very simple ROS wrapper.  That requires programming knowledge though.

closed 2025-05-12 by elecDeath