realsenseai/realsense-ros #3354: [D415] PointCloud generation fails at high resolutions with ROS2 Humble on Jetson Orin Nano

MauUrrutia, 2025-05-07:
**System Information**

- Device: Jetson Orin Nano
- JetPack Version: JetPack 6.2 (
- Kernel Version: 5.15.148-tegra
- ROS Distribution: Humble
- RealSense ROS Version: 4.55.1-1jammy.20250326.015335
- librealsense Version: 2.55.1
- Firmware: 5.16.0.1

**Expected Behavior**
When running `ros2 launch realsense2_camera rs_pointcloud_launch.py`, I expect:
1. Both depth and RGB modules to work simultaneously at high resolutions (e.g., 1280x720)
2. Smooth PointCloud generation with color mapping
3. Performance comparable to realsense-viewer operation

**Actual Behavior**
1. PointCloud generation freezes/stutters severely
2. Only depth module works reliably at higher resolutions
3. RGB module fails or drops frames unless resolution is drastically reduced
4. Performance is significantly worse than native realsense-viewer operation

**Steps to Reproduce**

- Install RealSense following Jetson-specific instructions: [`installation_jetson.md`](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md)
- Verify realsense-viewer works properly: `realsense-viewer`

- Install RealSense wrapper for ros2: `sudo apt install ros-humble-realsense2-*`
- Launch ROS2 wrapper: `ros2 launch realsense2_camera rs_pointcloud_launch.py accelerate_gpu_with_glsl:=true initial_reset:=true`
- Observe PointCloud issues: 

> $ ros2 launch realsense2_camera rs_pointcloud_launch.py accelerate_gpu_with_glsl:=true initial_reset:=true
[INFO] [launch]: All log files can be found below /home/mantecada01/.ros/log/2025-05-06-18-24-31-629993-mantecada01-desktop-12247
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [12248]
[INFO] [rviz2-2]: process started with pid [12250]
[realsense2_camera_node-1] [INFO] [1746577472.133935502] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1746577472.134328335] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1746577472.134366607] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1746577472.176906154] [camera.camera]: Device with serial number 021322060549 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1746577472.177164651] [camera.camera]: Device with physical ID /sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1746577472.177202315] [camera.camera]: Device with name Intel RealSense D415 was found.
[realsense2_camera_node-1] [INFO] [1746577472.177681963] [camera.camera]: Device with port number 2-1.3 was found.
[realsense2_camera_node-1] [INFO] [1746577472.177732172] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1746577472.177762476] [camera.camera]: Resetting device...
[rviz2-2] [INFO] [1746577472.648784892] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1746577472.649311901] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-2] [INFO] [1746577472.696407775] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1746577472.987518421] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1746577473.011679799] [rviz2]: Stereo is NOT SUPPORTED
[realsense2_camera_node-1] [INFO] [1746577478.460768579] [camera.camera]: Device with serial number 021322060549 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1746577478.461076803] [camera.camera]: Device with physical ID /sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1746577478.461152836] [camera.camera]: Device with name Intel RealSense D415 was found.
[realsense2_camera_node-1] [INFO] [1746577478.461912581] [camera.camera]: Device with port number 2-1.3 was found.
[realsense2_camera_node-1] [INFO] [1746577478.461992517] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1746577478.462304230] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1746577478.464394537] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1746577478.464493802] [camera.camera]: Device Name: Intel RealSense D415
[realsense2_camera_node-1] [INFO] [1746577478.464540938] [camera.camera]: Device Serial No: 021322060549
[realsense2_camera_node-1] [INFO] [1746577478.464576362] [camera.camera]: Device physical port: /sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1746577478.464610506] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1746577478.464640586] [camera.camera]: Device Product ID: 0x0AD3
[realsense2_camera_node-1] [INFO] [1746577478.464666922] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1746577479.693474454] [camera.camera]: Set ROS param depth_module.depth_profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1746577479.694420568] [camera.camera]: Set ROS param depth_module.infra_profile to default: 1280x720x30
[realsense2_camera_node-1] [WARN] [1746577479.706034765] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1746577479.714572861] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1746577479.723446061] [camera.camera]: Stopping Sensor: Depth Module
[... 55 lines cut ...]
[realsense2_camera_node-1] [WARN] [1746576617.667845859] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [WARN] [1746576617.676351378] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1746576617.685119691] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1746576617.685287826] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1746576617.692895517] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1746576617.703984146] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 480, Height: 270, FPS: 15
[realsense2_camera_node-1] [INFO] [1746576617.730351680] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1746576617.746107852] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 15
[realsense2_camera_node-1] [INFO] [1746576617.751278808] [camera.camera]: RealSense Node Is Up!

[image]

**Request:**
would greatly appreciate guidance on:

**Installation Validation**
Are there any known compatibility issues between:

- JetPack 6.2 (L4T R36.2)
- RealSense ROS2 Humble wrapper
- D415 cameras

Is there any misunderstanding on:
- Memory/GPU configuration for Orin Nano for the ros2 warpper
- Camera parameters for stable high-res PointCloud

MartyG-RealSense, 2025-05-07:
Hi @MauUrrutia  What is the performance like if you generate the pointcloud using the launch instruction below, please?

`ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true depth_module.depth_profile:=480x270x15 depth_module.infra_profile:=480x270x15 rgb_camera.color_profile:=480x270x15`

You do not need to enable align_depth in the launch command unless you need the aligned topics published, as the depth and RGB color topics are automatically mapped together to produce a textured pointcloud when depth and color are enabled.

MartyG-RealSense, 2025-05-07:
As you have a Jetson, I would recommend enabling librealsense's CUDA support if you have not done so already as this will provide similar acceleration of pointcloud processing to that of **accelerate_gpu_with_glsl** by offloading the work from the CPU to the Jetson's Nvidia GPU.

When using the Jetson instructions, the CUDA support should be enabled automatically if librealsense is installed from DKMS packages.  Otherwise, if you built librealsense from source code with CMake then the CUDA support is enabled by including the flag **-DBUILD_WITH_CUDA:=on** in the CMake build instruction.

********

However, I note that you installed the ROS wrapper using `sudo apt install ros-humble-realsense2-*` so this may be why **accelerate_gpu_with_glsl** was not working and the percentage usage of the CPU was high.  To use this acceleration feature, it is necessary to build the ROS wrapper from source code with 'colcon build' using the command below.

`colcon build --cmake-args '-DBUILD_ACCELERATE_GPU_WITH_GLSL=ON'`

MauUrrutia, 2025-05-07:
**Response to GPU Acceleration Suggestions**

Thank you for the guidance. I've implemented your recommendations:

1. Rebuilt librealsense with CUDA:
   ```bash
   cmake .. -DBUILD_WITH_CUDA=ON 

2. Compiled ROS wrapper with GLSL acceleration:

```bash
colcon build --cmake-args '-DBUILD_ACCELERATE_GPU_WITH_GLSL=ON'
```
Results:
✔ CPU usage dropped from ~80% to ~30% 
✔ Smoother pointcloud generation with accelerate_gpu_with_glsl:=true

**Follow-up Questions**

1. Optimal Configuration for D415

What would you recommend as the ideal profile settings for D415 on Jetson Orin Nano when using pointcloud? Specifically:
What's the maximum stable resolution for depth_module.depth_profile and rgb_camera.color_profile?
Should I enable or disable infra_profile when using textured pointclouds?
Is there a preferred FPS/resolution combination?

2. RViz Performance Issues

I'm experiencing significant FPS drops in RViz when displaying the pointcloud. Could you advise:
Are there specific RViz settings (QoS, Decimation) to improve performance?
Does the pointcloud message format affect rendering performance?

3. GPU Acceleration Verification
Despite enabling accelerate_gpu_with_glsl and seeing CPU improvement:
Why doesn't nvidia-smi show any GPU utilization?
How can I definitively confirm GPU acceleration is working?

**_Additional Context_**
Rviz and system monitor while ```ros2 launch realsense2_camera rs_launch.py``` is on:

[image]

[image]

MartyG-RealSense, 2025-05-08:
1.  It is difficult to suggest a profile for a specific computer.  I would recommend starting at 640x480 at 30 FPS for both depth and color profiles and if it is stable then increase to 1280x720 at 30 FPS and see if it remains stable.

1280x720 is the default resolution on the D415 camera model and is the optimal resolution for depth measurement accuracy.  1280x720 is also D415's default resolution for color.

If you do not define profiles then the D415's default resolution / FPS of 1280x720 at 30 FPS should be applied to both depth and color streams.

It is not necessary to have the infrared streams enabled or set an infrared profile if you are generating a depth pointcloud that is textured with RGB color.

2.  Applying decimation can improve performance because it 'downsamples' the resolution of the depth image to reduce its complexity.  It is common for users of a decimation filter to set its value to '2' so that the resolution is reduced by half (e.g 1280x720 with a decimation of 2 reduces the image resolution quality to 640x360).

3.  The main way to confirm if GPU acceleration is being applied is if there is a significant drop in CPU percentage usage compared to when GPU acceleration was not enabled.

MauUrrutia, 2025-05-08:
@MartyG-RealSense Just wanted to say thanks for your help! Your guidance on the GPU acceleration and optimal settings made a big difference. The pointcloud is running much smoother now.

Really appreciate you taking the time to explain:

The whole GLSL vs CUDA thing on Jetson

Those ideal resolution/FPS combos for the D415

Learned a lot through this issu

MartyG-RealSense, 2025-05-08:
You are very welcome and I'm pleased that I could help.  It's great to hear that you have achieved good pointcloud performance.  Thanks very much for the update!

closed 2025-05-15 by MauUrrutia