realsenseai/realsense-ros #3355: [D435i] Dual camera does not work

JasperTan97, 2025-05-07:
I am on the docker ros2 humble image, and my computer is on kernel 6.8.0-rt8 (realtime), and I have having problems launching 2 D435i camera together. I used `multi_camera_launch.py` but I get the following errors.

```
[realsense2_camera_node-2]  07/05 16:00:16,414 ERROR [138442591675968] (backend-v4l2.cpp:2970) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 40 error: No such device
[realsense2_camera_node-2]  07/05 16:00:16,414 ERROR [138442591675968] (backend-v4l2.cpp:2970) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 40 error: No such device
[realsense2_camera_node-2]  07/05 16:00:16,414 ERROR [138442591675968] (backend-v4l2.cpp:2970) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 40 error: No such device
[realsense2_camera_node-2]  07/05 16:00:16,414 ERROR [138442591675968] (backend-v4l2.cpp:2970) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 40 error: No such device
[realsense2_camera_node-2]  07/05 16:00:16,414 ERROR [138442591675968] (backend-v4l2.cpp:2970) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 40 error: No such device
[realsense2_camera_node-2]  07/05 16:00:16,414 ERROR [138443186894400] (librealsense-exception.h:52) xioctl(VIDIOC_STREAMOFF) failed for buf_type=13 Last Error: No such device
[realsense2_camera_node-2] [INFO] [1746633616.425132395] [camera2.D455]: Close Sensor - Done. 
[realsense2_camera_node-2]  07/05 16:00:16,613 ERROR [138442793002560] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[realsense2_camera_node-2]  07/05 16:00:16,614 ERROR [138442793002560] (error-handling.cpp:99) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[realsense2_camera_node-1]  07/05 16:00:16,688 ERROR [124624197097024] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
[realsense2_camera_node-1]  07/05 16:00:16,688 ERROR [124624197097024] (uvc-sensor.cpp:412) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
[realsense2_camera_node-1]  07/05 16:00:16,688 ERROR [124624197097024] (error-handling.cpp:99) Error during polling error handler: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
[realsense2_camera_node-1]  07/05 16:00:16,689 ERROR [124624230667840] (rs.cpp:237) [rs2_get_device_info( dev:nullptr, info:Product Id ) UNKNOWN] null pointer passed for argument "dev"
[realsense2_camera_node-1] [ERROR] [1746633616.689432920] [camera1.D435]: Error starting device: null pointer passed for argument "dev"
[realsense2_camera_node-1] [INFO] [1746633616.711490349] [camera1.D435]: Device with serial number 310622075022 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1746633616.711517065] [camera1.D435]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9.2/2-9.2:1.0/video4linux/video10 was found.
[realsense2_camera_node-1] [INFO] [1746633616.711541268] [camera1.D435]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1746633616.711630604] [camera1.D435]: Device with port number 2-9.2 was found.
[realsense2_camera_node-1] [ERROR] [1746633616.711635273] [camera1.D435]: The requested device with serial number 313522072909 is NOT found. Will Try again.
[realsense2_camera_node-1] [INFO] [1746633617.070551606] [camera1.D435]: Checking new devices...
[realsense2_camera_node-2] [INFO] [1746633617.070822671] [camera2.D455]: Checking new devices...
[realsense2_camera_node-2] [INFO] [1746633617.099506246] [camera2.D455]: Device with serial number 313522072909 was found.
[realsense2_camera_node-2] 
[realsense2_camera_node-2] [INFO] [1746633617.099542121] [camera2.D455]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9.1/2-9.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-2] [INFO] [1746633617.099547147] [camera2.D455]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-2] [INFO] [1746633617.099632412] [camera2.D455]: Device with port number 2-9.1 was found.
[realsense2_camera_node-2] [INFO] [1746633617.099637142] [camera2.D455]: Device USB type: 3.2
[realsense2_camera_node-2] [INFO] [1746633617.099650706] [camera2.D455]: getParameters...
[realsense2_camera_node-1] [INFO] [1746633617.099684208] [camera1.D435]: Device with serial number 313522072909 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1746633617.099717408] [camera1.D435]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9.1/2-9.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1746633617.099724134] [camera1.D435]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1746633617.099833213] [camera1.D435]: Device with port number 2-9.1 was found.
[realsense2_camera_node-1] [INFO] [1746633617.099843727] [camera1.D435]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1746633617.099858498] [camera1.D435]: getParameters...
[realsense2_camera_node-2] [INFO] [1746633617.100013674] [camera2.D455]: JSON file is not provided
[realsense2_camera_node-2] [INFO] [1746633617.100036352] [camera2.D455]: Device Name: Intel RealSense D435I
[realsense2_camera_node-2] [INFO] [1746633617.100048337] [camera2.D455]: Device Serial No: 313522072909
[realsense2_camera_node-2] [INFO] [1746633617.100056935] [camera2.D455]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9.1/2-9.1:1.0/video4linux/video0
[realsense2_camera_node-2] [INFO] [1746633617.100066394] [camera2.D455]: Device FW version: 5.13.0.55
[realsense2_camera_node-2] [INFO] [1746633617.100074943] [camera2.D455]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1746633617.100036101] [camera1.D435]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1746633617.100040995] [camera1.D435]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1746633617.100048368] [camera1.D435]: Device Serial No: 313522072909
[realsense2_camera_node-1] [INFO] [1746633617.100052305] [camera1.D435]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9.1/2-9.1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1746633617.100055763] [camera1.D435]: Device FW version: 5.13.0.55
[realsense2_camera_node-1] [INFO] [1746633617.100058833] [camera1.D435]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1746633617.100061840] [camera1.D435]: Sync Mode: Off
[realsense2_camera_node-2] [INFO] [1746633617.100128899] [camera2.D455]: Sync Mode: Off
[realsense2_camera_node-2] [INFO] [1746633617.253423825] [camera2.D455]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1746633617.254049643] [camera1.D435]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-2] [INFO] [1746633617.254304385] [camera2.D455]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1746633617.254841564] [camera1.D435]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-2] [INFO] [1746633617.264746286] [camera2.D455]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1746633617.265623012] [camera1.D435]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-2] [INFO] [1746633617.266342129] [camera2.D455]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-2] [INFO] [1746633617.266480067] [camera2.D455]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1746633617.267174720] [camera1.D435]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1746633617.267397728] [camera1.D435]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1746633617.272850620] [camera1.D435]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1746633617.272930215] [camera1.D435]: Stopping Sensor: RGB Camera
[realsense2_camera_node-2] [INFO] [1746633617.273370619] [camera2.D455]: Stopping Sensor: Depth Module
[realsense2_camera_node-2] [INFO] [1746633617.273544604] [camera2.D455]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1746633617.274166916] [camera1.D435]: Starting Sensor: Depth Module
[... cut ...]
I can't find the first error when it fails, because the ` (backend-v4l2.cpp:2970) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 40 error: No such device` error spams my terminal and locks it so I can't kill the terminal fast enough. 

However, if I use `ExecuteProcess` or run 2 rs_launch on 2 different terminals, everything works fine. 

I have 2 questions then. The more urgent is, would there be any invisible problems, since currently it seems to work fine on Rviz2? The second is what I can do to go back to use the multi_camera_launch. 

Thanks.

MartyG-RealSense, 2025-05-08:
Hi @JasperTan97  Do either of the multi-camera launch instructions below work for you, please?

`ros2 launch realsense2_camera rs_multi_camera_launch.py camera_name1:=d435i_1 device_type1:=d435i camera_name2:=d435i_2 device_type2:=d435i`

`ros2 launch realsense2_camera rs_multi_camera_launch.py serial_no1:=_313522072909 serial_no2:=_310622075022`

MartyG-RealSense, 2025-05-09:
These commands were adapted from the documentation for the previous-generation 'ros2-legacy' RealSense ROS2 wrapper, the naming commands in the first launch command may not now be suited for the current wrapper.  I'm pleased to hear that the command based on serial numbers works for you though.

closed 2025-05-14 by JasperTan97