realsenseai/realsense-ros #3358: Issue with Node Dying After USB Reconnection in D455 Camera with `enable_pointcloud=true`

TrashKim, 2025-05-15:
We are using the D455 camera with ROS Melodic, and the ROS package version is 2.3.2.

In the launch file for the camera, we are using `enable_pointcloud=true`.
However, the USB connection of the D455 camera was temporarily disconnected and reconnected (approximately 0.3 seconds), and after generating the following messages, the node dies (this also happens if the USB cable is forcibly disconnected and reconnected).

```
 15/05 16:50:38,808 WARNING [140635639703296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
 15/05 16:50:38,808 ERROR [140635390592768] (ds5-thermal-monitor.cpp:94) Error during thermal compensation handling: usb device disconnected
[ERROR] [1747295438.835445037]: The device has been disconnected!
terminate called without an active exception
[camera/realsense2_camera_manager-2] process has died [pid 26864, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/twinny/.ros/log/371f8bf8-3161-11f0-8114-04ea5681ad30/camera-realsense2_camera_manager-2.log].
log file: /home/twinny/.ros/log/371f8bf8-3161-11f0-8114-04ea5681ad30/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/twinny/.ros/log/371f8bf8-3161-11f0-8114-04ea5681ad30/camera-realsense2_camera-3*.log
```

If we set `enable_pointcloud=false`, even when I forcibly disconnect and reconnect the USB cable, the node does not die, and it successfully reconnects.
Additionally, both image and depth data are received without any issues.

```
 15/05 16:48:35,706 WARNING [140378155104000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
 15/05 16:48:35,706 ERROR [140378171889408] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: usb device disconnected
[ERROR] [1747295315.748588919]: The device has been disconnected!
 15/05 16:48:35,754 WARNING [140377306687232] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 82, error: No such device, number: 13
 15/05 16:48:35,766 WARNING [140377557227264] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 84, error: No such device, number: 13
[ERROR] [1747295315.857998118]: Exception: failed to set power state
 15/05 16:48:35,857 ERROR [140378155104000] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 15/05 16:48:35,857 ERROR [140378205460224] (sensor.cpp:572) acquire_power failed: failed to set power state
 15/05 16:48:36,285 ERROR [140378155104000] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 15/05 16:48:36,285 ERROR [140377650624256] (sensor.cpp:572) acquire_power failed: failed to set power state
 15/05 16:48:36,285 ERROR [140377650624256] (error-handling.cpp:78) Error during polling error handler: failed to set power state
[ WARN] [1747295316.884096407]: No RealSense devices were found!
[ INFO] [1747295322.912341497]:  
 15/05 16:48:42,983 WARNING [140378163496704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1747295323.073384435]: Device with serial number 241122303813 was found.
[ INFO] [1747295323.073433029]: Device with physical ID 2-1-14 was found.
[ INFO] [1747295323.073456822]: Device with name Intel RealSense D455 was found.
[ INFO] [1747295323.073800902]: Device with port number 2-1 was found.
[ INFO] [1747295323.073830049]: Device USB type: 3.2
[ INFO] [1747295323.075667788]: getParameters...
[ INFO] [1747295323.129482497]: setupDevice...
[ INFO] [1747295323.129515541]: JSON file is not provided
[ INFO] [1747295323.129535127]: ROS Node Namespace: camera
[ INFO] [1747295323.129557670]: Device Name: Intel RealSense D455
[ INFO] [1747295323.129580191]: Device Serial No: 241122303813
[ INFO] [1747295323.129602208]: Device physical port: 2-1-14
[ INFO] [1747295323.129624564]: Device FW version: 05.13.00.55
[ INFO] [1747295323.129655707]: Device Product ID: 0x0B5C
[ INFO] [1747295323.129683607]: Enable PointCloud: Off
[ INFO] [1747295323.129708798]: Align Depth: On
[ INFO] [1747295323.129733859]: Sync Mode: On
[ INFO] [1747295323.129771392]: Device Sensors: 
[ INFO] [1747295323.224164407]: Stereo Module was found.
[ INFO] [1747295323.248340642]: RGB Camera was found.
[ INFO] [1747295323.248503145]: Motion Module was found.
[ INFO] [1747295323.248521940]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
```

The launch file we are using is as follows (We are using the default `rs_camera.launch`):

```
<launch>
  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="depth_width" value="424"/>
    <arg name="depth_height" value="240"/>
    <arg name="color_width" value="1280"/>
    <arg name="color_height" value="720"/>
    <arg name="depth_fps" value="30"/>
    <arg name="color_fps" value="30"/>
    <arg name="align_depth" value="true"/>
    <arg name="enable_pointcloud" value="true"/>
    <!-- <arg name="enable_accel" value="true"/> -->
    <arg name="clip_distance" value="3"/>
    <!-- <arg name="filters" value="temporal,decimation,spatial,disparity"/> -->
  </include>
</launch> 
```

The camera is connected to the robot via USB, and when the robot moves quickly, there is a very low-frequency but occasional USB communication disturbance. Because of this, the issue has been reported at the site where the robot is being operated.

I would appreciate any assistance with this issue.

MartyG-RealSense, 2025-05-24:
Hi @TrashKim  Do you require further assistance with this case, please?  Thanks!

TrashKim, 2025-06-02:
Sorry for the late reply.
We were running some tests, so we got back to you a bit late.
Using `rs_rgbd.launch` solved the reconnection issue.
Thank you for your help!

MartyG-RealSense, 2025-06-02:
You are very welcome, @TrashKim - I'm pleased to hear that your issue is solved.  Thanks very much for the update!