realsenseai/realsense-ros #3360: External Trigger (inter_cam_sync_mode=4) Works Only on D455F, Not on D435iF or D455

madhu-dhanasekaran, 2025-05-16:
Hi,

I'm working with multiple RealSense cameras and trying to use external hardware triggering using:

**inter_cam_sync_mode:=4**

This mode works only on the D455f. However, when I apply the same configuration to my D435if and D455, they do not respond to external trigger pulses.

**Setup**
Model	Firmware	Result
D455F	5.16.0.1	✅ External trigger works
D455	5.16.0.1	❌ No trigger response
D435iF	5.16.0.1	❌ No trigger response

External trigger pulses are verified and wired to the sync input pin using the 9 pin connector.

RealSense ROS2 driver launched with inter_cam_sync_mode:=4

Running on Ubuntu & ROS2 Humble

**Clarification Needed**

  Does the D435if & D455 support inter_cam_sync_mode:=4?

rs-enumerate-devices -c doesn't show "Inter Cam Sync Mode" at all for this device.

realsense-viewer shows the mode dropdown, but setting mode 4 has no effect.

Is the D455f the only RealSense model with full external hardware trigger support (mode 4)?

**Goal**

I need to externally trigger multiple RealSense cameras at 30 FPS using GPIO signals for frame-level synchronization. Currently, only D455f is responding as expected.

Any clarification on hardware/firmware support differences between D455F, D455, and D435iF is appreciated.

Let me know if you'd like to include diagnostic command outputs or full ROS2 launch configs as well.

MartyG-RealSense, 2025-05-17:
Hi @madhu-dhanasekaran   Yes, you should be able to use genlock mode 4 with D435if and D455.  The filter should not make a difference to the process.

If you are using an external trigger pulse with mode 4 then the FPS of the camera should be 2x the frequency of the trigger.  So if your trigger's frequency is 30 FPS then the camera should be set to 60 FPS.  And if you want the camera to use 30 FPS then the trigger frequency should be 15.

60 FPS should be supported by the camera at 848x480 depth resolution.  1280x720 resolution only supports up to 30 FPS.

madhu-dhanasekaran, 2025-05-19:
Hi @MartyG-RealSense

Thank you very much for your reply. Now, I adjusted FPS of the camera to be 2x the trigger pulse frequency and I see that it is waiting for the trigger. But, however I saw a weird behavior. The camera is waiting for an external trigger until the 9 pin connector is connected to it. But as soon as the connector is connected, it receives frames even when the trigger pulse is not sent. I mean the other end of the 9 pin connector is open. This was quite weird. I observe this only with D435if & D455 and D455f still works as expected i.e., the exposure starts when it is triggered and stops when the trigger is stopped. 

By the way can you also confirm if 3.3 V pulse should be fine for the external trigger or should it be strictly 1.8V?

Thank you.

MartyG-RealSense, 2025-05-19:
The trigger pulse should be 1.8 V in order to be compatible with the sync pin (Pin 5).  If your trigger is not 1.8 V then you can alter the voltage using an electronics component known as a voltage shifter or level shifter.  RealSense users have used a level shifter similar to the one at the link below.

https://www.sparkfun.com/products/12009

I have not heard of the frame generation behaviour that you describe.  Can you confirm please that on your sync cable's 9-pin connector there are only two wires attached to it, going to Pin 5 (Sync) and Pin 9 (Ground).

[image]

madhu-dhanasekaran, 2025-05-19:
I have all 9 wires connected, however they are open on the other end except for 5 and 9. I can remove other cables.

Also, I would like to confirm one more aspect. By 1.8V you mean it works for the 5V logic? Does it mean anything below 1.8V is considered as logic 0 and above as 1?

MartyG-RealSense, 2025-05-30:
Hi @madhu-dhanasekaran  After you earlier achieved a solution, do you require further assistance with this case please?  Thanks!

MartyG-RealSense, 2025-05-30:
Thanks very much for the confirmation of a solution.  I will close this case.  Thanks again!