realsenseai/realsense-ros #3362: Unable to launch two cameras (D435 & D455) and publish RGB, depth, and pointcloud at the same time

madhu-dhanasekaran, 2025-05-22:
I am trying to enable hardware trigger on different realsense depth cameras and succeeded in achieving it individually. Now I want to connect all cameras and trigger them simultaneously. As I went through other issues I tried to launch rs_launch.py on 2 different terminals with different camera_namespace and camera_name. But, when the second camera is launched, it interferes with the first one and both does not publish properly. When one of them is stopped, the other starts to publish.

Can you please guide me to publish all variants for both the cameras simultaneously? For setting up the camera in hw trigger mode, I have to set the inter_cam_sync_mode param to 4 or above and for doing this I have to start publishing with enable_depth:=false. When I do this I get

[realsense2_camera_node-1] [ERROR] [1747924024.059623249] [topic_2.cam_2]: ./src/rs_node_setup.cpp:487:An exception has been thrown: xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy

However, I saw one strange behavior that both the cameras try to access the same camera node "realsense2_camera_node-1". So, it is clear that rs_launch.py can not be used for my use case.

Steps tried:
1. Terminal1: ros2 launch realsense2_camera rs_launch.py camera_namespace:=topic_1 camera_name:=cam_1
    Terminal2: ros2 launch realsense2_camera rs_launch.py camera_namespace:=topic_2 camera_name:=cam_2
2. Installed librealsense SDK with -DFORCE_RSUSB_BACKEND=TRUE flag and invoked 
    ros2 launch realsense2_camera rs_multi_camera_launch.py
    I get,
    [realsense2_camera_node-1]  22/05 16:29:35,951 ERROR [134655915308608] (backend-v4l2.cpp:2970) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 21 error: No such device

MartyG-RealSense, 2025-05-22:
Hi @madhu-dhanasekaran  Does launch with the multi-camera launch file work if you provide the serial number of the two cameras in the launch instruction?  For example:

`ros2 launch realsense2_camera rs_multi_camera_launch.py serial_no1:=_123456789 serial_no2:=_987654321`

Substitute the serial numbers of your own two cameras into the above instruction, and make sure to put the underscore symbol ( _ ) before the serial number.

madhu-dhanasekaran, 2025-05-22:
Hi @MartyG-RealSense This helped. Thanks. However, can I also publish pointcloud in the same way? I do not see any topics for that in rviz2. 

Also, can you tell me how I can set the inter_cam_sync_mode param before launching the launch file? Thanks.

MartyG-RealSense, 2025-05-22:
The pointcloud is enabled and published by adding to the launch instruction `pointcloud.enable:=true`

Do you mean that you do not wish to set `inter_cam_sync_mode:=4` in the launch instruction like in your earlier issue at https://github.com/IntelRealSense/realsense-ros/issues/3360 please?

If so then you can set ROS2 parameters in a **.yaml** configuration file that imports and applies the launch parameters defined within it, as described at https://github.com/IntelRealSense/realsense-ros/issues/1640

MartyG-RealSense, 2025-05-22:
My knowledge of yaml files is limited, unfortunately.

The most relevant reference I could find was a RealSense ROS user at https://github.com/IntelRealSense/realsense-ros/issues/3137 who took the approach of editing the rs_launch.py launch file to import the yaml settings into it, and then used the rs_multi_camera_launch.py multi-camera launch file for their actual launch (since the multicam launch file imports the settings of rs_launch.py via the `import rs_launch` command).

madhu-dhanasekaran, 2025-05-23:
Thanks for pointing to #3137. I could now enable pointcloud as well for both the cameras. Now I wanted to add a third camera and I tried extending the rs_multi_camera_launch. py launch file by adding `camera_name3` and `camera_namespace3` to the local_parameters and subsequent lines in `generate_launch_description` function to include a third camera. But, I found this is not sufficient to detect a third cam and work with it. Can you please let me know if I am missing something here? Thanks

MartyG-RealSense, 2025-05-23:
The multicam launch file is intended for two cameras.  I have not previously seen a case where it has successfully been adapted for three cameras.

After your modifications to the launch file, did you add a **serial_no3** parameter to the end of your launch instruction with the serial number for the third camera, please?

madhu-dhanasekaran, 2025-05-23:
Yes, I have made the **serial_no3** modification as well. 

It would be really helpful if you can guide me to adapt this launch file for adding the third one as well :-)

MartyG-RealSense, 2025-05-23:
A workaround that does not require having to adapt the file might be to open a second terminal and perform an `rs_launch.py `launch with the serial number specified so that it does not try to access the other two cameras that are on the multicam launch file in the other terminal.  For example, again substituting your own serial number in:

`ros2 launch realsense2_camera rs_launch.py serial_no:=_123456789 pointcloud.enable:=true`

*********

If I was going to adapt the multicam launch file myself then I would do it in the way below, removing the static TF part of the file that calculates a transform between two cameras.  You are welcome to try the launch file below, though I do not have the means to test it myself to see if the approach works.

```
# Copyright 2023 Intel Corporation. All Rights Reserved.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

# DESCRIPTION #
# ----------- #
# Use this launch file to launch 3 devices.
# The Parameters available for definition in the command line for each camera are described in rs_launch.configurable_parameters
# For each device, the parameter name was changed to include an index.
# For example: to set camera_name for device1 set parameter camera_name1.
# command line example:
# ros2 launch realsense2_camera rs_multi_camera_launch.py camera_name1:=D400 device_type2:=l5. device_type1:=d4..

"""Launch realsense2_camera node."""
import copy
from launch import LaunchDescription, LaunchContext
import launch_ros.actions
from launch.actions import IncludeLaunchDescription, OpaqueFunction
from launch.substitutions import LaunchConfiguration, ThisLaunchFileDir
from launch.launch_description_sources import PythonLaunchDescriptionSource
import sys
import pathlib
sys.path.append(str(pathlib.Path(__file__).parent.absolute()))
import rs_launch

local_parameters = [{'name': 'camera_name1', 'default': 'camera1', 'description': 'camera1 unique name'},
                    {'name': 'camera_name2', 'default': 'camera2', 'description': 'camera2 unique name'},
       {'name': 'camera_name3', 'default': 'camera3', 'description': 'camera3 unique name'},
                    {'name': 'camera_namespace1', 'default': 'camera1', 'description': 'camera1 namespace'},
                    {'name': 'camera_namespace2', 'default': 'camera2', 'description': 'camera2 namespace'},
{'name': 'camera_namespace3', 'default': 'camera3', 'description': 'camera3 namespace'},
                    ]

def set_configurable_parameters(local_params):
    return dict([(param['original_name'], LaunchConfiguration(param['name'])) for param in local_params])

def duplicate_params(general_params, posix):
    local_params = copy.deepcopy(general_params)
    for param in local_params:
        param['original_name'] = param['name']
        param['name'] += posix
    return local_params

def generate_launch_description():
    params1 = duplicate_params(rs_launch.configurable_parameters, '1')
    params2 = duplicate_params(rs_launch.configurable_parameters, '2')
    params3 = duplicate_params(rs_launch.configurable_parameters, '3')
    return LaunchDescription(
        rs_launch.declare_configurable_parameters(local_parameters) +
        rs_launch.declare_configurable_parameters(params1) +
        rs_launch.declare_configurable_parameters(params2) +
    rs_launch.declare_configurable_parameters(params3) +
        [
        OpaqueFunction(function=rs_launch.launch_setup,
                       kwargs = {'params'           : set_configurable_parameters(params1),
                                 'param_name_suffix': '1'}),
        OpaqueFunction(function=rs_launch.launch_setup,
                       kwargs = {'params'           : set_configurable_parameters(params2),
                                 'param_name_suffix': '2'}),
OpaqueFunction(function=rs_launch.launch_setup,
                       kwargs = {'params'           : set_configurable_parameters(params3),
                                 'param_name_suffix': '3'}),
    ])
```

madhu-dhanasekaran, 2025-05-23:
Thanks a lot @MartyG-RealSense I will try this and let you know :-)

madhu-dhanasekaran, 2025-05-23:
Hi @MartyG-RealSense 

The idea to use the original `rs_multi_camera_launch.py` along with a `rs_launch.py` in a different terminal worked. However, I feel it would be a better and a neat solution to have a single launch file.

And the changes you made were the exact changes I made to the `rs_multi_camera_launch.py` except for removing the static TF part. However both our versions did not help unfortunately.

MartyG-RealSense, 2025-05-30:
Hi @madhu-dhanasekaran  Do you require further assistance with this case, please?  Thanks!

madhu-dhanasekaran, 2025-05-30:
Hi @MartyG-RealSense Yes I would require further assistance with this, thanks. The above mentioned solution did not work exactly in my case.

MartyG-RealSense, 2025-05-30:
I researched this issue from the beginning again but unfortunately did not identify a better solution than using a combination of the multicam and single-camera launch scripts, tried earlier at https://github.com/IntelRealSense/realsense-ros/issues/3362#issuecomment-2904724358

madhu-dhanasekaran, 2025-05-30:
Okay, then I would recommend to close this issue for now.

MartyG-RealSense, 2025-05-30:
Okay, thank you very much.  I will close the issue as you recommend.

closed 2025-05-30 by MartyG-RealSense