realsenseai/realsense-ros #3363: Unable to use L515 for multi-camera hardware synchronization

ZJU-ZongYue, 2025-05-25:
I wired according to the white paper. When I plugged in the cable of the sync clock setting port and turned on hardware synchronization in the SDK, sometimes the camera's radar started flashing and sometimes not. 
Which situation is right? Did I not plug the cable in tightly?

And two of the three cables are required. How can I determine the order of the two cables (from left to right or other?)
Hope to get your answer as soon as possible

ZJU-ZongYue, 2025-05-25:
This is my wiring
[image]

MartyG-RealSense, 2025-05-25:
Hi @ZJU-ZongYue  May I first confirm that you are basing your sync cable's design on the L515 hardware sync guide, which has different wiring from the one for 400 Series camera models?

https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup#1-connecting-the-cameras

[image]

MartyG-RealSense, 2025-05-25:
In regard to the order of the cables, if you orient your L515 to the same position as the one in the image below then Pin 1 should be the one that corresponds to the red cable.

[image]

MartyG-RealSense, 2025-05-25:
In L515 sync, the Inter Cam Sync Mode of the cameras should be set to '1'.  Are yours set to 1 in the Viewer?

Also, how many L515 are you using?  Hardware sync of L515 is not necessary when there is only one camera, because there are not any other L515 cameras present to cause interference and it provides no benefit for a single camera as it does not synchronize timestamps like the 400 Series version of sync does.

ZJU-ZongYue, 2025-05-28:
I used two L515 cameras to capture the same object, set the Inter Cam Sync Mode to "1" and connected it to the Raspberry Pi. The Raspberry Pi pins provided differential signals of high and low levels, but the radar lights of both cameras remained constantly on and there was interference in the imaging.
Could you please explain the reason for this? Normally, should the radar lights be flashing?

ZJU-ZongYue, 2025-05-28:
Thank you for your reply. I may have misrepresented myself earlier. 
My two L515 cameras are connected to my i5-13600KF desktop computer, with the Raspberry Pi 4B only serving as a synchronization interface for the high and low level trigger signals, as indicated in the white paper.
Should this configuration be acceptable?

closed 2025-06-15 by MartyG-RealSense