realsenseai/realsense-ros #3366: Missing debian ROS package for arm and jazzy

dvdmc, 2025-06-01:
I am glad that I managed to install `librealsense` from ROS debian packages with `ros-jazzy-librealsense2*` on an old Jetson Nano (ARM) directly! 🙏🏼 However, there is no such package for `sudo apt install ros-jazzy-realsense2-*`. I have checked that the repository compiles without issues from source on the Jetson Nano.

This issue is to add, if possible, binary debian ROS packages for Jazzy ARM support. Also, I would recommend changing the README `Step 2` in relation to "Jetson users" since it seems that only the Option 1 is available while Option 2 works flawlessly and is easier to set up.

MartyG-RealSense, 2025-06-01:
Hi @dvdmc   I have highlighted your question to my Intel RealSense colleagues on the ROS team.

Nir-Az, 2025-06-01:
Hi @dvdmc ,
according to the ROS2 packages status `ros-jazzy-realsense2-*` packages exist.
See here: https://repo.ros2.org/status_page/ros_humble_default.html?q=realsense

Could be an apt-get update is needed?
Which JetPack are you using on your Jetson that is based on Ubuntu 24 (ROS2 Jazzy)?

Latest JetPack 6.2 is based on Ubuntu 22 IMO.
Could be you need to try ROS2 Humble?

dvdmc, 2025-06-01:
Thanks @MartyG-RealSense and @Nir-Az for your prompt response!

@Nir-Az, I'm using JetPack 4.6.6 [L4T 32.7.6], which is quite old. Then I am running ROS Jazzy inside a Docker, which works with no problems.

I'm not sure how to interpret `repo.ros2.org` but does it also covers `arm64` builds?

I pretty sure it was not an update issue and, as I mentioned, I managed to install the `arm64` package for `ros-jazzy-librealsense2*` just before trying `sudo apt install ros-jazzy-realsense2-*`. Since it failed, I compiled it from source and had no problem.

dvdmc, 2025-06-01:
I checked the correct status page: https://repo.ros2.org/status_page/ros_jazzy_unv8.html?q=realsense and it seems to exist. Unless you can confirm it doesn't, let me check tomorrow ECT the reason why I couldn't find it :)

On the other hand, changing the README if packages can be installed on jetson from ROS repositories would be nice for Jetson users.

dvdmc, 2025-06-02:
Great! I will close this for now and reopen if I have any issues. I will also try a PR with changes of the install instructions for Jetson. Thanks a lot @Nir-Az @MartyG-RealSense

closed 2025-06-02 by dvdmc

MartyG-RealSense, 2025-06-02:
@dvdmc  Thanks very much for the update.  This is not an issue that I can check myself, unfortunately.  @Nir-Az will hopefully be able to assist you further when they have the opportunity to do so.

dvdmc, 2025-06-18:
Sorry for the late response, I am currently on travel. As I mentioned a while ago (the message is above), I am using JetPack 4.6.6 [L4T 32.7.6], which is quite old. Then I am running ROS Jazzy inside a Docker, which works with no problems. The Docker pulls from `FROM arm64v8/ros:jazzy-ros-base`. The Dockerfile is from [this repo](https://github.com/dvdmc/drones_as2_unizar/blob/main/docker-nano/Dockerfile).

Nir-Az, 2025-06-22:
I understand, 
I am not sure why ROS2 bloom release does not handle this automatically.
We will investigate it, but as it seems for now build from source is needed for the RS ROS Camera node but for liberalness you can install the Debian's from the installation guide

MartyG-RealSense, 2025-07-17:
Thanks very much, @dvdmc - I'm pleased to hear that you achieved a working setup via compiling.  Let's leave this case open for a while longer in case anybody else wants to add a comment.