realsenseai/realsense-ros #3377: Missing aligned depth topics

Eku127, 2025-06-13:
System: Ubuntu 22.04
ROS: ROS Humble
Branch: ros2-master

Hi! Iwant to get the aligned depth image output for mapping, I found out the missing of the `aligned_depth_to_color ` topics. Any suggestions? Thanks!

[image]

MartyG-RealSense, 2025-06-13:
You are very welcome!

Unless you need the aligned topics to be published, it is not necessary to have align_depth enabled in order to generate a depth pointcloud that has RGB (a 'textured' pointcloud).  If you add `pointcloud.enable:=true` to your launch instruction then if depth and color are both enabled (which they are by default), the pointcloud will automatically map depth and color together.

RealSense cameras output real-world depth measurements in meters by default, so you do not need to adjust the depth values.  The ROS wrapper has a Python 'node script' called **show_center_depth.py** that will use the ROS topics to print this information when the script is run after the ROS wrapper has been launched.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/realsense2_camera/scripts/show_center_depth.py

Eku127, 2025-06-13:
Thanks again! I’ll give this a try and really appreciate the prompt support!