realsenseai/realsense-ros #3393: 435i ROS2 Jazzy Compatibility

LevinTamir, 2025-07-10:
I'm trying to understand if I can migrate my project from ROS 2 Humble to Jazzy while continuing to use the Intel RealSense 435i, which currently works with my setup on Humble. Are there any known limitations or compatibility issues with the RealSense 435i on Jazzy and Ubuntu 24.04?

MartyG-RealSense, 2025-07-10:
Hi @LevinTamir  Support for Jazzy on Ubuntu 24.04 was added to the RealSense ROS2 wrapper in August 2024 in version 4.56.1 onwards (the current wrapper version at the time of writing this is 4.56.3).  I have not seen reports of compatibility or limitation problems since then.

closed 2025-07-21 by MartyG-RealSense

LevinTamir, 2025-07-26:
Hi,

I'm currently trying to use the realsense2_camera package with ROS 2 Jazzy on Ubuntu 24.04 for an Intel RealSense D435i setup. Despite following the instructions and using the latest ros2-development branch from the realsense-ros repository, I'm encountering build errors related to:

rs2::rotation_filter not being found

rs2::motion_frame missing get_combined_motion_data()

rs2::auto_calibrated_device missing get_calibration_config() and set_calibration_config()

Here’s a screenshot of the full build error:

[image]

MartyG-RealSense, 2025-07-26:
Hi @LevinTamir  Which version of the librealsense SDK did you install please, and what method did you use to install it?

Also, do you know which version of the camera's firmware driver is currently installed in the camera?

kaleido-jean, 2026-01-20:
Hi, I have the same issue to colcon build realsense2_camera for jazzy. The system is Ubuntu 24.04, and I installed the SDK2 following this [link](https://github.com/realsenseai/librealsense/blob/master/doc/installation.md#troubleshooting-installation-and-patch-related-issues) since my kernel is 6.14. When I colcon build the packages, it shows this:

[image]