realsenseai/realsense-ros #3395: ROS2 filter parameter support

guest43, 2025-07-16:
I am using D435 with ROS2 jazzy and realsense 2.56.3. I would like to add realsense2 filters to the depth camera stream. Since they cannot be applied with the realsense API after the topics are already published, they need to be applied before the camera publishes to the topics. However, i cannot find documentation that points to how to use the realsense filters (spatial..temporal..decimation..threshold...etc) in a ROS2 launch file. 

I see that others have had a similar [issue ](https://github.com/IntelRealSense/realsense-ros/issues/583), and it was resolved by using dynamic_reconfigure nodes, but dynamic_reconfigure was replaced in ROS 2. If I understand correctly, the launch parameter features of ROS 2 cover what used to be done with dynamic_reconfigure.  In rs_launch.py, the configurable parameters include many of the filters (although the disparity filter is ambiguous and the threshold/clamping filter is missing), but only ability to enable/disable and not set the values of the parameters.

Additionally, filters are not exported in the json settings file of the realsense-viewer, so calling the json config file doesn't appear to be an option either.

What am I missing? Is there a way to specify filters and their values in a ROS 2 launch file?

MartyG-RealSense, 2025-07-16:
Hi @guest43  https://github.com/IntelRealSense/realsense-ros/issues/2921 has information about enabling and configuring the post-processing filters and their parameters in the RealSense ROS2 wrapper.  In the ROS2 wrapper, filters are enabled with the format `<filter name>enable:=true>` - for example:

`ros2 launch realsense2_camera rs_launch.py temporal_filter.enable:=true`

You can then configure the settings of the filter after launch with a **ros2 param set** command.

guest43, 2025-07-16:
Hey @MartyG-RealSense, thanks for the response! That looks like what I need.. can you link to where these parameters are documented? The parameter names look like they might be a little different than those in the python sdk. If I use fake parameter names in the launch file, it doesn't throw any errors so I'm not sure how to figure out if the parameters are correctly being implemented.
(1) Can you confirm if this syntax is correct?
```
'spatial_filter.enable': 'true',
'spatial_filter.hole_filling': 'X',
'spatial_filter.magnitude': 'X',
'spatial_filter.smooth_delta': 'X',
'spatial_filter.smooth_alpha': 'X',
'decimation_filter.enable': 'true',
'decimation_filter.magnitude': 'X',
'temporal_filter.enable': 'true',
'temporal_filter.smooth_alpha': 'X',
'temporal_filter.smooth_delta': 'X',
'temporal_filter.persistence': 'X',
'disparity_filter.enable': 'true',
'disparity_filter.transform_to_disparity': 'true',
'hole_filling_filter.enable': 'true',
'hole_filling_filter.mode': 'X'
```
 (2) Also, are the filters implemented in the order in which they are specified in the launch file, or is there a default ordering in the underlying routines?

guest43, 2025-07-16:
ah, okay `ros2 param list /<camera_node>` gives me the syntax for all available filters.

Thanks for the recommendation on post-processing filters and performance.

closed 2025-07-16 by guest43