realsenseai/realsense-ros #3396: Redundant test dependencies in package.xml of realsense2_description

Plumezz, 2025-07-17:
Hello, after building and testing this project with branch ros2-master, I found something unexpected.

I noticed the `package.xml` file in the `realsense2_description` package contains the following content which means a format check.
```
  <test_depend>ament_lint_auto</test_depend>
  <test_depend>ament_lint_common</test_depend>
```
But there are no test section in `CMakeLists.txt` resulting in no testing programs to run actually.

So I think if we want the ament_lint tools to do a format check, maybe need add the testing section in the `CMakeLists.txt` like following:
```
if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  ament_lint_auto_find_test_dependencies()
endif()
```
After adding the content, amen_lint_tools like `ament_copyright, ament_lint_cmake, ament_xmllint, ament_flake8, ament_pep257`,  can be executed correctly when I use `colcon test` command.

But there are some format errors after testing. Here is the result file:

[realsense2_description_result.log](https://github.com/user-attachments/files/21283326/realsense2_description_result.log) 
2:flake8  3:lint_cmake  4:pep257

If we do not need the ament_lint_tools, I think the test_depend might be a redundant dependency that could be removed to reduce the burden of project maintenance.

I would appreciate it if you could tell me how do you think of this?

Thanks!

MartyG-RealSense, 2025-07-17:
Hi @Plumezz  I will highlight your question to my RealSense colleagues on the ROS development team.  Thanks!

Nir-Az, 2025-07-23:
@ashrafk93 can you take a look?
Seems like a valid point, maybe we can drop this demand?

Plumezz, 2025-07-26:
Ok, thanks!