realsenseai/realsense-ros #3402: D435i has no image under ros

WALYP, 2025-07-25:
I have screen under realsense-viewer, but when I start the camera under ros, it gives me the error: WARNING [140074092054272] (messenger-libusb.cpp:42) control_transfer returned error, index: 300 , error: Resource temporarily unavailable, number: b. And there is no screen under rviz, can anyone tell me how to fix this.

[image]

[image]

MartyG-RealSense, 2025-07-25:
Hi @WALYP  Is realsense-viewer closed when you launch in ROS?  The Viewer tool is not required to publish topics in the RealSense ROS wrapper.

Does it make a difference if you set your Fixed Frame in RViz to **camera_link** instead of base_link?

I note that you are using the new librealsense SDK version 2.56.4 published 3 days ago.  There is not a ROS wrapper published for use with 2.56.4 at the time of writing this.  ROS wrappers for new librealsense versions often follow within days of the librealsense release.  At the moment the latest librealsense version supported by a wrapper is 2.56.3 (for wrapper 4.56.3).

Sometime an older wrapper version can be used with a newer librealsense.  However, librealsense 2.56.4 has extensive changes, including how ROS is handled, in order to support the new-generation RealSense D555 camera that is coming soon.

WALYP, 2025-07-26:
Thank you for your reply. I am aware of this issue, and now that I have installed librealsensev2.54.2 and realsense-ros, it is working. In v2.54.2, the Gyro stream pointer kept moving erratically, but this does not happen in v2.56.4. The Accel stream is normal, and I have calibrated the IMU. Also, my kernel version is too high, so I can't get accurate timestamps.

[image]

[image]

MartyG-RealSense, 2025-07-26:
In 2.55.1 onwards, a dynamic gyro sensitivity system was introduced that requires firmware 5.16.0.1 or newer.  If you are using 5.16.0.1 or newer with an SDK older than 2.55.1 (such as 2.54.2) then gyro values may be corrupted.  If you will be using 2.54.2 then the solution to the corruption would be to downgrade your firmware version to **5.15.1.0**, the firmware designed for use with 2.54.2.

https://dev.realsenseai.com/docs/firmware-releases-d400#d400-series-firmware-downloads

[image]

If librealsense is built from source code with the **libuvc backend** installation method then hardware timestamps are supported without having to apply a patch script to the kernel.