realsenseai/realsense-ros #3412: Some problems with realsense_ros on D435i

jin-yiwei, 2025-08-08:
I used one computer to control three cameras（D435i），it's fine, now i control six cameras, new camera will report warning like this: 
[image]
i want to know if i control too many cameras so the bandwidth is not enough? or i make other mistakes?
Here are my terminal comman:
`ros2 launch realsense2_camera rs_launch.py camera_namespace:='xxx' camera_name:='left_camera' align_depth.enable:=true initial_reset:=true serial_no:=\"'327122078556'\" diagnostics_period:=0.2`

`ros2 launch realsense2_camera rs_launch.py camera_namespace:='xxx' camera_name:='middle_camera' align_depth.enable:=true initial_reset:=true serial_no:=\"'327122078556'\" diagnostics_period:=0.2`

`ros2 launch realsense2_camera rs_launch.py camera_namespace:='xxx' camera_name:='right_camera' align_depth.enable:=true initial_reset:=true serial_no:=\"'327122078556'\" diagnostics_period:=0.2`

Here are my librealsense & realsense_ros version:
> RealSense ROS v4.55.1
Built with LibRealSense v2.55.1
Running with LibRealSense v2.55.1

MartyG-RealSense, 2025-08-08:
Hi @jin-yiwei  Using alignment is processing-intensive as it is calculated on the computer's CPU and not on the camera hardware.  So it is conceivable that going from alignment with three cameras to six computers could result in a frames timeout due to the increased processing burden.

If an Nvidia Jetson was used as the computer then the processing burden could be offloaded from the CPU onto the graphics GPU using the SDK's support of CUDA.  The librealsense SDK has an alternative to CUDA of offloading called GLSL for non-Jetson computers that be used to accelerate alignment.  However, the wrapper documentation about enabling GLSL in the ROS wrapper lists support for accelerating pointclouds and the colorizer but not alignment.

Unless there is a problem that is preventing the cameras from launching when not using initial_reset then I would recommend leaving initial_reset out of your launch command in order to simplify the launch process.

It looks as though the launch is using the default stream profile of 848x480 depth and 1280x720 color for all cameras.  That could certainly cause bandwidth to be used up if repeated for 6 cameras.  Please try adding the command below to your launch instruction to reduce bandwidth by setting both depth and color to 640x480 at 15 FPS in order to test if insufficient bandwidth is a cause of your problem.

`depth_module.depth_profile:=640x480x15 rgb_camera.color_profile:=640x480x15`

jin-yiwei, 2025-08-11:
> If an Nvidia Jetson was used as the computer then the processing burden could be offloaded from the CPU onto the graphics GPU using the SDK's support of CUDA.
Only jeston can use CUDA accelerating? I'm using NVIDIA 4090, could this can accelerate by CUDA?  And GLSL it's not support for alignment, right?
And I will try low resolution, thank you for your advice!

MartyG-RealSense, 2025-08-11:
CUDA requires a computer equipped with Nvidia graphics, whether built-in or with a  graphics card, as it does not work with non-Nvidia brands of graphics chip.  

Theoretically a PC equipped with Nvidia graphics could make use of CUDA support.  99% of RealSense users who use CUDA have an Nvidia-equipped Jetson though, so it is very rare for someone to try getting CUDA support working on a PC.

MartyG-RealSense, 2025-08-11:
The initial_reset has a similar effect to unplugging and re-inserting the camera in the USB port, or using the Hardware Reset option in the 'More' menu of the realsense-viewer tool.  The camera disconnects and then attempts to reconnect.  However, in rare cases on some computers the camera may not automatically reconnect after the disconnection.  So if you can avoid using initial_reset then it's one less potential problem during the launch process.