realsenseai/realsense-ros #3416: [ROS2 realsense D455 Jetson  ]    [camera.camera]: Device 1/1 failed with exception: bad optional access

yong2015789, 2025-08-13:
platform: ROS2 humble ;Jetson AGX Orin; realsense D455
run :ros2 run realsense2_camera    realsense2_camera_node

error: as follows:
13/08 11:50:22,197 WARNING [281472662288608] (ds-motion-common.cpp:452) No HID info provided, IMU is disabled
 13/08 11:50:22,198 ERROR [281472662288608] (rs.cpp:256) [rs2_create_device( info_list:0xffff60020f70, index:0 ) UNKNOWN] bad optional access
 13/08 11:50:22,198 ERROR [281472662288608] (rs.cpp:256) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
 13/08 11:50:22,198 WARNING [281472662288608] (rs.cpp:392) null pointer passed for argument "device"
[WARN] [1755057022.198784515] [camera.camera]: Device 1/1 failed with exception: bad optional access
[ERROR] [1755057022.198895739] [camera.camera]: The requested device with  is NOT found. Will Try again.

how to slove the above error? Thank you!

yong2015789, 2025-08-13:
@MartyG-RealSense   the ablove issue can't slove!

yong2015789, 2025-08-13:
@MartyG-RealSense  how to  downgraded to a firmware?   
now  firmware of d455  is 5.16.0.1

yong2015789, 2025-08-13:
version of the librealsense SDK is V2.26.5
when I update firmware , open software of realsense-viewer ,
it can't display image and depth data

MartyG-RealSense, 2025-08-13:
If you have the master release then your version number is likely to be 2.56.5 rather than 2.26 (which was released in the year 2019).

The recommended firmware to use with 2.56.5 is the latest 5.17.0.10 version.

MartyG-RealSense, 2025-08-13:
@yong2015789  Did you update the firmware to 5.17.0.10 please?  If you did, is the camera working now?