realsenseai/realsense-ros #3423: External tigger for intel realsense d455 for time sync

bathirappan-k, 2025-08-19:
I am trying to trigger the d455 using external trigger from arduino, I cross verified the pulses from arduino using oscilloscope, it is exactly 60hz with 50% duty cycle and Vmax is close 4v. I am using pin 5 for trigger and pin 9 gnd. I am using below given parameters file to launch the camera, and i am using jetson orin nx developer board, with ros2 humble in docker. but still when ever i echo the topics i get only 55 to 57 hz and how to cross verify my trigger is working properly or not. how can i verify my data is generated based on default launch file or based on external trigger as with and without trigger i get same hz while ros2 echo of camera raw image topic. first of all is my connect is right or not based on the picture provided? thanks in advance for the support.

launch file:

import os
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, IncludeLaunchDescription, LogInfo, OpaqueFunction
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node
from ament_index_python.packages import get_package_share_directory

def launch_setup(context, *args, **kwargs):

    rs_config = LaunchConfiguration('rs_config').perform(context)

    debug_config = LogInfo(msg=[
        f'Using RealSense config file: {rs_config}'
    ])
    
    bringup_share_dir = get_package_share_directory('realsense2_bringup')
    camera_share_dir = get_package_share_directory('realsense2_camera')
    rs_launch = IncludeLaunchDescription(
    PythonLaunchDescriptionSource([
        os.path.join(camera_share_dir, 'launch'),
        '/rs_launch.py'
    ]),
    launch_arguments={'config_file': os.path.join(bringup_share_dir, 'config', rs_config)}.items())

    return [
        debug_config,
        rs_launch,
    ]

def generate_launch_description():

    rs_config_arg = DeclareLaunchArgument(
        'rs_config',
        default_value='rgbd_inertial.yaml',
        description='Path to config file')
    
    opaque_launch = OpaqueFunction(function=launch_setup)

    return LaunchDescription([
        rs_config_arg,
        opaque_launch
    ])

params file:

inital_reset: true
enable_sync: true

# RGB
enable_color: true
rgb_camera.color_profile: 640x480x60
#rgb_camera.enable_auto_exposure: false

# Depth
enable_depth: true
depth_module.depth_profile: 640x480x60
align_depth.enable: true
clip_distance: 2.5

# IRs
enable_infra1: true
enable_infra2: true
depth_module.infra_profile: 640x480x60
depth_module.emitter_enabled: 0

# IMU
enable_gyro: true
enable_accel: true
gyro_fps: 400
accel_fps: 200
unite_imu_method: 2
hold_back_imu_for_frames: true

# External Trigger Settings - CRITICAL for sync
inter_cam_sync_mode: 1  
sync_mode: 1   

[image]

MartyG-RealSense, 2025-08-30:
Hi @bathirappan-k  Do you require further assistance with this case, please?  Thanks!

bathirappan-k, 2025-09-09:
It is taking longer time than expected for time sync with another sensor, as my other sensor can use ppt, gps, pps. I was trying pps, but implementing ppt in other sensor is easy for me, i just want to know does intel realsense d455 has ptp time sync, if so, how to activate it and use it, if any docs available please share here. 

I also tried output_trigger_enabled, which generates 1.8v, 50microseond pulse but my other sensors requires 20-200milliseconds and it uses pps, so only one pulse at the start, is there any possible way to do this by changing parameters in the config file. 

as my high priority if for ptp time sync, any support related to that is much appreciated.

MartyG-RealSense, 2025-09-09:
As far as I am aware, RealSense cameras do not have built-in PTP support.

The link below has a guide from a person using a RealSense D415 who attempted to **simulate** PTP synchronization in software.

https://vcl3d.github.io/VolumetricCapture/docs/synchronization/

bathirappan-k, 2025-09-25:
1. if set `global_time_enabled: true` does it take time from system or intel realsense internal clock? 
2. I want the camera to use the system time to which device it is connected to. so which parameter should I enable or disable?
3. if both `global_time_enabled: true` and `inter_cam_sync_mode: 2` does the camera acts a slave and still take the time from the system it is connected or even without `global_time_enabled: true` the camera is going to take time from the connected system?

this is a bit confusing, when ros2 wrapper is used from which source the timestamps of ros2 msgs are taken from? from internal realsense or the connected system? if 30 hz is set in the parameter file where does all the time calculation takes place in the internal realsense clock or from the system?

MartyG-RealSense, 2025-09-25:
1.  Global Time compares the device time (the camera clock) and the system time (the computer clock), as described at https://github.com/IntelRealSense/librealsense/pull/3909

2.  The only way to use system time would be to build the librealsense SDK from source code with CMake with libuvc backend set to False, and not apply a patch script to the kernel.  By doing so, support for hardware timestamps in the SDK will not be enabled and the SDK will provide 'software' timestamps based on the system clock.

3.  If hardware timestamp support is not enabled in the SDK then Global Time will be inactive and the timestamps will be based on system time. 

In regard to the RealSense ROS wrapper, its timestamps will be based on whatever timestamping type is active in the librealsense SDK.  If hardware timestamps are enabled then the ROS wrapper will use those, and if hardware timestamps are unavailable then the system timestamps will be used.

bathirappan-k, 2025-09-28:
could you please elaborate more about how to build it? do you have any documentation to follow? 

here is how I build in docker container 

the below section is build before installation of ros2 humble

libuvc_installation.sh

```
#!/bin/bash -e

apt-get update

rm -rf ./librealsense_build
mkdir librealsense_build && cd librealsense_build

echo "Installing Librealsense-required dev packages"
apt-get install git cmake usbutils libssl-dev freeglut3-dev libusb-1.0-0-dev pkg-config libgtk-3-dev unzip -y

rm -f ./master.zip
wget https://github.com/IntelRealSense/librealsense/archive/master.zip
unzip ./master.zip -d .
cd ./librealsense-master

echo "Install udev-rules"
cp config/99-realsense-libusb.rules /etc/udev/rules.d/ 
cp config/99-realsense-d4xx-mipi-dfu.rules /etc/udev/rules.d/
udevadm control --reload-rules && udevadm trigger 

mkdir build && cd build
cmake ../ -DFORCE_RSUSB_BACKEND=on -DCMAKE_BUILD_TYPE=release -DBUILD_WITH_CUDA=on
make -j2
make install

echo -e "\e[92m\n\e[1mLibrealsense script completed.\n\e[0m"
```

then using this in docker file to build like

```
# Install librealsense from source
COPY libuvc_installation.sh /usr/local/bin/
WORKDIR /usr/local/bin
RUN chmod +x libuvc_installation.sh && ./libuvc_installation.sh 
```

then commands to install ros2 humble 

ros2 wrapper for realsense 

```
# Install realsense-ros from source
RUN rosdep init && rosdep update

WORKDIR /home/realsense_ws/src
RUN git clone https://github.com/IntelRealSense/realsense-ros.git -b 4.55.1

WORKDIR /home/realsense_ws
RUN apt-get update && \
    rosdep install -i --from-path src --rosdistro humble --skip-keys=librealsense2 -y

RUN . /opt/ros/humble/setup.sh && \
    colcon build

RUN echo "source /home/realsense_ws/install/local_setup.bash" >> ~/.bashrc
```

then given section is used to hard dependency for `ros-humble-librealsense2` so creating a dummy package to satisfy the requirements and use the build from source.

```
RUN echo "Section: misc\nPriority: optional\nStandards-Version: 3.9.2\n\nPackage: ros-humble-librealsense2\nVersion: 2.55.1\nArchitecture: arm64\nMaintainer: Dummy <dummy@dummy.com>\nDescription: Dummy package for ros-humble-librealsense2" > ros-humble-librealsense2.equivs && \
    equivs-build ros-humble-librealsense2.equivs && \
    dpkg -i ros-humble-librealsense2_2.55.1_arm64.deb
```

in the above mentioned could you please let me know what needs to be changed for using
1) system timestamps what need to changed
2) hardware trigger what need to changed

sharing both is much appreciated as I am trying multiple methods to perform time sync, not sure which is going to work 

apart the above 

3) i have a assumption there a few frame drops due to computational load in the jetson, so I tried setting 10 fps for all the output mode, but it automatically setting to random fps and ros2 topic is 20hz to 27 hz why this is happening? is there are fixed constrains for fps with respective to resolution or any other fixed constrains to use fps for rgb, depth and stereo. if there are any documentation to follow this could you please it here.

4) is possible to use system time based on the trigger from external embedded system? as when i implement ptp for lidar it is time synced with the system i am using, so if my realsense takes time from system i am connected to there will be very less difference in time in between lidar and camera. if the external trigger is used by camera and take time from the system it is connected to, then there will be precise time sync for camera and lidar.

5)if hardware trigger is sent the timestamp is taken from camera or the system where the camera is connected to?

I really appreciate your time for answering all these questions. thanks.

MartyG-RealSense, 2025-10-02:
10 FPS is not a supported selectable speed on RealSense cameras.  The supported speeds are 5, 15, 30, 60 and 90.  If you attempted to set a stream configuration in ROS that used 10 FPS then the configuration would be rejected and the default stream configuration would be applied instead.

[image]

It is possible to configure a custom frequency in ROS by defining a **ROS Rate**.

https://roboticsbackend.com/ros-rate-roscpy-roscpp/

*************

If librealsense was installed from DKMS packages then you can uninstall all RealSense packages on a computer by using the command below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

IF you built librealsense from source code and then built the ROS wrapper afterwards, to uninstall librealsense you would have to go to the **build** directory of the librealsense source code folder and run the CMake command below:

`sudo make uninstall && make clean`

I believe that if both librealsense and the ROS wrapper were installed from the ROS servers with **sudo apt install** commands then simply completely deleting the entire **/ros2_ws/src/** catkin workspace folder will be sufficient to uninstall from the computer.

**********

You used to be able to check if a kernel had a librealsense patch applied to it (meaning that it would have hardware timestamps enabled) by running the command below.  If the output did not contain the string 'realsense' then the kernel was unpatched.

`modinfo uvcvideo | grep "version:"`

From around kernel 6.5 onwards though, the output did not contain the word 'realsense' but it was still patched, so the method would not work for recent kernel versions.

A way to guarantee that the kernel is  unpatched would be to install a different kernel version, as the newly installed kernel will be unpatched and so software timestamps would be provided.

MartyG-RealSense, 2025-10-22:
Okay, no problem - I look forward to your next report.  I will reopen the issue in the meantime.  Good luck!

rtarun1, 2025-10-26:
I understand this thread is focused on hardware synchronization. However, I’m specifically interested in synchronizing the RGB streams between multiple D455 cameras. Since I don’t have access to trigger cables, what’s the best way to achieve software-based synchronization?

MartyG-RealSense, 2025-10-26:
Hi @rtarun1  RealSense 400 Series cameras such as D455 have a feature called **Global Time** enabled by default that generates a common timestamp for all streams between multiple cameras attached to the same computer, as described at https://github.com/IntelRealSense/librealsense/pull/3909

If you are able to program in C++ language and you only need to take a single frame capture from all cameras, you could consider using a script at https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412899468 that captures an RGB image from all attached cameras simultaneously.

rtarun1, 2025-10-26:
Hi @MartyG-RealSense , thank you for the response.

Ideally, I’d like to obtain a synchronized 30 FPS stream, for which I can write a basic ROS wrapper. I also noticed that the ROS wrapper doesn’t currently expose the global_time_enabled parameter.

So, just to confirm — the right approach would be to extend the provided code to capture a synchronized video stream, and then build a ROS wrapper to publish it as ROS messages. Is that correct?

Atleast for now I don't need a syncronized real-time output. I just need all the cameras to have a common UNIX time. So i can sync them offline

MartyG-RealSense, 2025-10-26:
Global Time is enabled by default in the librealsense SDK that runs in the background underneath the ROS wrapper layer, so you should not need to enable it in ROS.  

Global Time being enabled is not as important in ROS as it is in librealsense though as ROS has its own built-in time correction mechanism, which makes Global Time redundant.

You can certainly write your own simple ROS wrapper to use instead of the official RealSense wrapper and a few other RealSense ROS users have done this.  As far as I am aware though, ROS should already be time-synching the topics from different cameras without having to write your own mechanism to do so,