realsenseai/realsense-ros #3425: High latency issue when two subscribers listen to RealSense camera topics

algo-scope, 2025-08-21:
### Description

I am publishing RealSense topics on **ROS 2 Foxy**. My camera node configuration is as follows:

```yaml
realsense_params:
  enable_color: true
  enable_depth: true
  depth_module.enable_auto_exposure: true
  rgb_camera.enable_auto_exposure: true
  pointcloud.enable: false
  align_depth.enable: false
  serial_no: "241122300333"
  depth_module.profile: "640x480x15"
  rgb_camera.profile: "640x480x15"
  enable_gyro: false
  enable_accel: false
  color_qos: 'SENSOR_DATA'
  depth_qos: 'SENSOR_DATA'
  enable_sync: false
  global_time_enabled: false
```

Both of my subscriber nodes are configured with the following QoS:

```python
sensor_qos = QoSProfile(
    reliability=ReliabilityPolicy.BEST_EFFORT,
    durability=DurabilityPolicy.VOLATILE,
    history=HistoryPolicy.KEEP_LAST,
    depth=1
)
```

I am using **CycloneDDS** as the RMW implementation.

### Expected Behavior

When subscribing with two nodes at the same time, the messages should still be delivered with a delay close to the camera frame interval (around 0.06–0.1s for 15 FPS).

### Actual Behavior

As soon as **two subscriber nodes are running**, the delay between the message timestamp and the reception time becomes more than **1 second**.
Even if both subscriber callbacks do **nothing** (empty callbacks with no processing), the delay still occurs.

### Additional Information

* ROS 2 distro: **Foxy**
* Camera: **RealSense D455**
* RMW: **CycloneDDS**

MartyG-RealSense, 2025-08-21:
Hi @algo-scope  Which librealsense SDK and RealSense ROS2 wrapper versions are you using, please?  Support in the wrapper for Foxy ended from wrapper version 4.54.1 onwards.

Also, have you tried changing depth_qos and color_qos to QoS modes other than SENSOR_DATA please?  Other supported QoS modes are:

DEFAULT
SYSTEM_DEFAULT
PARAMETER_EVENTS
SERVICES_DEFAULT
PARAMETERS

Does the delay reduce if you lower the FPS from 15 to 5 (the minimum FPS supported on D455) ?

Does it make a difference if you change your Reliability Policy setting from BEST_EFFORT to RELIABLE

youtudou, 2025-08-21:
Hi, I am facing a similar issue. When I set the color_qos to SENSOR_DATA mode, my subscriber fps is significantly lowered. I have tested a few sets of parameters:
1. color profile: 640x480x30, qos: SYSTEM_DEFAULT qos, I can get 30fps on my subscriber
2. color profile: 640x480x30, qos: SENSOR_DATA, I can only get a varying 1-20 frames, and the ros2 topic hz shows around 7.
3. color profile: 1280x720x30, qos: SENSOR_DATA, I can only get a varying 1-20 frames, and the ros2 topic hz shows around 7.
4. color profile: 424x240x30, qos: SENSOR_DATA, stable 30fps
5. color profile: 424x240x60, qos: SENSOR_DATA, stable 60fps

Could you please let me know how could I get a stable subscriber fps similar to the publisher fps if I want to stick with color profile: 640x480x30, qos: SENSOR_DATA

MartyG-RealSense, 2025-08-22:
As Foxy is End of Life (EOL) it is no longer supported by the RealSense ROS wrapper.  So if 4.55.1 is being used, it is difficult to know whether the problem is being caused by the absence of official Foxy support and whether the issue would be occurring in more recent ROS2 versions such as Humble or Iron.

Are the two nodes on the same computer or separate computers?  Some years ago a RealSense ROS user reported that their two nodes worked on the same computer but they had problems when the nodes were on separate computers. 

https://robotics.stackexchange.com/questions/92301/ros2-realsense2-subscribing-to-raw-camera-from-different-computer-throws-excep

algo-scope, 2025-08-25:
I've changed to v4.51.1 but it still had 0.8s delay. The two nodes are on the same computer with same ROS_DOMAIN_ID

MartyG-RealSense, 2025-08-25:
When I was working on the QoS case of @youtudou at https://github.com/IntelRealSense/realsense-ros/issues/3427 I learned at https://github.com/IntelRealSense/realsense-ros/issues/3427#issuecomment-3214580135 that the SENSOR_DATA QoS profile has a small sized queue and that SYSTEM_DEFAULT - which "uses the RMW implementation’s default values for all of the policies" - might be a more suitable choice.

MartyG-RealSense, 2025-09-10:
Okay, thanks very much for the update.  Please do let me know if you make progress during your investigations.